ARMmbed/mbed-os #5939: get_tick assert fails

drahnr, 2018-01-25:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug? / Question
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
NRF52DK

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-os sha:**
5f6572179 (tag: mbed_lib_rev157, tag: mbed-os-5.6.6, origin/mbed-os-5.6) Merge pull request #5634 from ARMmbed/release-candidate

**DAPLink version:**

**Expected behavior**
Nop

**Actual behavior**
mbed assertation failed: os_timer->get_tick() == svcRtxKernelGetTickCount(), file: ./mbed-os/rtos/TARGET_CORTEX/mbed_rtx_idle.c

**Steps to reproduce**
Happens to me all the time as soon as I use a debugger, I would rather know if this is either a known issue or what can cause this.
My program works fine unless a debugger is attached.

drahnr, 2018-01-25:
Possibly related:
https://os.mbed.com/questions/79770/nRF52832-Assert-Lockup/
https://github.com/ARMmbed/mbed-os/pull/5595

0xc0170, 2018-01-25:
@c1728p9 Please review

c1728p9, 2018-01-25:
Halting the device in a debugger for too long or being in a critical section for too long can cause this assert to occur. This is not ideal so I created #5942 to remove the assert.