ARMmbed/mbed-os #5940: I2C->transfer callback

pilotak, 2018-01-25:
Hi i'm trying to implement I2C async function `transfer` to library but I can't compile due to 
```
[Error] test.cpp@5,19: invalid use of non-static member function 'void Test::internal_cb_handler(int)'
```

Here is source code:
`test.h`
```cpp
#include "mbed.h"

#define ADDRESS (0x17 << 1)

class Test {
  public:
    Test();
    void init(I2C * i2c_obj);
    void measure(uint8_t data, event_callback_t function);

  protected:
    I2C * i2c;
    event_callback_t done_cb;
    event_callback_t internal_cb;

    char tx_buf[1];
    char rx_buf[3];

    void internal_cb_handler(int event);
};
```
`test.cpp`
```cpp
#include "mbed.h"
#include "test.h"

Test::Test() {
    internal_cb = internal_cb_handler;
}

void Test::init(I2C * i2c_obj) {
    i2c = i2c_obj;
}

void Test::measure(uint8_t data, event_callback_t function) {
    done_cb = function;
    tx_buf[0] = data;

    if (i2c->transfer(ADDRESS, (char*)tx_buf, 1, (char*)rx_buf, 3, internal_cb, I2C_EVENT_ALL) != 0) {
        printf("busy\n");

    } else {
        printf("transfer OK\n");
    }
}

void Test::internal_cb_handler(int event) {
    if (event & I2C_EVENT_ERROR_NO_SLAVE) {
        printf("not responding\n");

    } else if (event & I2C_EVENT_ERROR) {
        printf("error\n");

    } else {
        printf("sending done\n");
        done_cb.call(123);
    }
}
```

`main.cpp`
```cpp
#include <mbed.h>
#include "test.h"

I2C i2c(PB_7, PB_6);
Test test;

void callback(int result) {
  printf("result: %i", result);
}

int main() {
  test.init(&i2c);

  while (true){
    test.measure(0b00010000, callback);
    Thread:wait(500);
  }
}
```
so when I change the `test.cpp@5` to `internal_cb.attach(this, &Test::internal_cb_handler);` it will compile but this is deprecated. What is the way to do it please? I didn't find any example except [this one](https://os.mbed.com/teams/SiliconLabs/code/SILABS_RHT/) where i got the solution from. Thank you.

### Description
- Type: Bug

**Target**
STM32

0xc0170, 2018-01-26:
I do not fully understand the need to capture the method to the variable. You dont need that one, can be removed (`internal_cb`). As error indicates `event_callback_t` is not the same as declaration for `internal_cb_handler`

You just need to pass your own internal_cb_handler to the `transfer` method

pilotak, 2018-01-26:
The thing is when transfer is done, i need to wait a few milisecods and then ask again if data is ready, if not wait again and when i have a result then pass is to main

0xc0170, 2018-01-29:
> The thing is when transfer is done, i need to wait a few milisecods and then ask again if data is ready, if not wait again and when i have a result then pass is to main

Based on my earlier comment, you should be able to fix the errors you are getting. Once it is, what is still missing in the application? You can illustrate it via a code snippet?

pilotak, 2018-01-30:
Here is flow diagram, as you can see all things are done in internal callback and when i have a valid data pass them to `done_cb` not earlier. All i'm asking is what to do in order to compile my example a posted, i will figure out the rest.

[image]

0xc0170, 2018-01-30:
> All i'm asking is what to do in order to compile my example a posted, i will figure out the rest.

Need to pass a proper callback signature, see my comment above, I'll quote it here:

> As error indicates event_callback_t is not the same as declaration for internal_cb_handler (former is a typedef Callback<void(int)> event_callback_t; , the latter is non-static member function)

pilotak, 2018-01-30:
Thank you for helping me. When i change declaration to `static void internal_cb_handler(int event);` that gives me an error about `done_cb`.
Could I please ask you to put changes you said into my test class? I'm really stuck here

0xc0170, 2018-02-01:
If you change one line, transfer? Not using internal internal_cb as it is not needed?

 ```
    if (i2c->transfer(ADDRESS, (char*)tx_buf, 1, (char*)rx_buf, 3, event_callback_t(this, &Test::internal_cb_handler), I2C_EVENT_ALL) != 0) {
```

pilotak, 2018-02-01:
Thank you very much i would not figure it out without you. I can now understand why `internal_cb` it not needed.

closed 2018-02-01 by pilotak