ARMmbed/mbed-os #5955: ticker unit test contains a race condition, never runs one of the callback events

drewcassidy, 2018-01-27:
### Description
- Type: Bug 
- Related issue: discovered in #5892
- Priority: Minor

The last test in the ticker unit tests appears to contain a race condition and never calls one of its callbacks. The overhead of the callbacks also appears to block the test from continuing.

I have no clue what the intended behavior of these callbacks is, but from my testing it appears to be doing the following
1) ticker is started for Event A every 1ms
2) Event A runs, starting another ticker for Event B every 1ms
3) 1ms later, Event A gets called again, and it resets the ticker for event B
4) this continues indefinitely

On the systems I've tested this on (nRF51822 both with and without a crystal for the LF clock) the length of the callbacks appears to be enough to completely block the main loop of the test program, causing the DUT to never respond to queries from the host. I am, however, using a non-standard board, so to be absolutely sure I will try the same tests on an nRF51-DK once I get hold of one

---------------------------------------------------------------
## Bug

**Target**
nRF51822

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.3.0

**Expected behavior**
Honestly I dont know the purpose of this test?

**Actual behavior**
`ticker_callback_2_switch_to_1` never gets called

**Steps to reproduce**
run `mbed test -c -v -n mbed-os-tests-mbed_drivers-ticker`

## Notes
This test should probably have some documentation as to its intended behavior. I can tell its trying to test the timer drift but other than that its unclear what the control flow should be between these callbacks

0xc0170, 2018-01-29:
@drewcassidy To be certain we are using the same sha - what is the mbed-os version or commit sha ?

@mprse @maciejbocianski Please review

drewcassidy, 2018-01-29:
> what is the mbed-os version or commit sha

this was at https://github.com/ARMmbed/mbed-os/commit/26d0c6dcafa66f1e52de2ba7ea320aa7cc68176a, but the file hasnt changed since

maciejbocianski, 2018-01-30:
fix is on the way

closed 2018-02-08 by drewcassidy