ARMmbed/mbed-os #5958: VS Code exporter shouldn't overwrite launch.json

amq, 2018-01-29:
### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

`launch.json` has to be modified by the user to match the debugger and target. But once e.g. a new libary is added and the project is re-exported to add the new includes, the file gets overwritten by the default one which is nothing more than a template.

**Suggested enhancement**

If `launch.json` exists, do not touch it.

**Pros**

The user-defined launch / debugging configuration persists over project exports.

**Cons**

Can't think of any.

0xc0170, 2018-01-29:
@theotherjimmy Please review

drahnr, 2018-01-30:
This behaviour is the same or all exporters, but tbh, it is rather annoying. It is not uncommon to have the projectfiles in the gitignores and as such this risks wiping out user configuration.
A warning and abort is the safer route, telling the user what is necessary to continue and what implications this has.

amq, 2018-01-30:
If I compare to Eclipse / `gnuarmeclipse`, the debug and build configuration is not overwritten (it's not created at all) and the project-specific settings are also not touched.

|          | Eclipse     | VS Code               |
|----------|-------------|-----------------------|
| Includes | .cproject   | c_cpp_properties.json |
| Build    | not created | tasks.json            |
| Debug    | not created | launch.json           |
| Settings | not created | settings.json         |

As a result, the Eclipse exporter has the behavior which I expect: only includes are re-exported. I am not saying that we should stop the VS Code exporter from generating those other files though, mainly because they are created with a nice GUI in Eclipse, while they would have to be created manually in VS Code.

theotherjimmy, 2018-01-30:
@drahnr When you say warning and abort, I got a bit confused.
I would imagine a warning would look something like:
```
warning: overwriting old .cproject file
```
and an abort looking like:
```
error: an export has already happened in this project; a .cproject file is already present. Please delete this file before exporting.
```
I would not expect to be able to do both simultaneously.

theotherjimmy, 2018-01-30:
@amq That's not quite correct. The GNU ARM Eclipse exporter generates a `.project` file next to a `.cproject` file. This includes the build configuration. (if it did not, how would an exported project build?). Debug settings and other settings may be overwritten if they are stored in the `.project` file.

amq, 2018-01-30:
@theotherjimmy ah, you're right. Never actually had to look into that file or change it. My main point is about `launch.json` and `settings.json` which are expected to be modified by the user.

drahnr, 2018-01-30:
@theotherjimmy I clearly missed something here and/or my test was flawed.

theotherjimmy, 2018-01-30:
@drahnr I was trying to point out that I see those 2 options: warn, abort; as mutually exclusive.

drahnr, 2018-01-30:
@theotherjimmy In this case I would go for `abort`, `warning+backup` would work to some extent, but repeated execution would cause the same issue again with the backup and clutter the project.

closed 2018-02-01 by cmonr