ARMmbed/mbed-os #5959: LittleFS - mkdir does not return error for existing directory

moranpeker, 2018-01-29:
Bug
Toolchain:
GCC_ARM|ARM|IAR

Actual behavior
please follow this mkdir scenario:
/
mkidr 1
cd 1
mkdir 2
/1/2
mkdir /1

mkdir /1 should return is exists error.

0xc0170, 2018-01-29:
@geky Please review

geky, 2018-01-29:
Hi @moranpeker, thanks for reporting the issue. Which compiler is this with?

Is this with the `chdir` function?

moranpeker, 2018-01-31:
with mkdir function

moranpeker, 2018-02-11:
Hi @geky , any update?

geky, 2018-02-11:
Sorry, no progress yet. Do you have a code snippet that reproduces this issue? That would help speed things up.

moranpeker, 2018-02-14:
lfs info: mkdir("pri/work1", 0x180)
lfs info: mkdir -> 0
lfs info: mkdir("pri/work1/pri/work1", 0x180)   _(relative path!)_
lfs info: mkdir -> -2

geky, 2018-02-15:
So, I just tested with all three toolchains (GCC, ARM, IAR), none of them actually have the function `chdir`. How are you changing the directory?

moranpeker, 2018-02-18:
the issue is about **mkdir** not chdir.

geky, 2018-02-18:
Looking at these calls:
```
mkdir("pri/work1", 0x180)
mkdir("pri/work1/pri/work1", 0x180)
```

The directory "pri/work1/pri" does not exist when you try to make the "pri/work1/pri/work1" directory. mkdir will only make one directory at a time, so it returns ENOENT (-2). This is correct behaviour.

I think I'm not understanding what the actual issue is. Are you able to post a code snippet that demonstrates the issue?

moranpeker, 2018-02-20:
**the scenario:**
1.make dir : "pri/work1"
2.try to make dir "pri/work1" (again!!! ) - i expect an "directory already exist " error and not try to creating "pri/work1/pri/work1" dir.

**actual behavior:**
lfs info: mkdir("pri/work1", 0x180)
lfs info: mkdir -> 0
lfs info: mkdir("pri/work1/pri/work1", 0x180) 
lfs info: mkdir -> -2

i dont expect this kind of behavior.
why does it try to create "pri/work1/pri/work1" directory ? if the path of : "pri/work1 " is already exist ?
i think the problem is about relative / absolute path.

geky, 2018-02-21:
Ok, I'm still not able to reproduce your results, but I think I tracked down the source of the problems. It looks like FilePath was left undefined when passed a path with no leading slashes. I've fixed that here so it should now follow the path correctly: https://github.com/ARMmbed/mbed-os/pull/6156.

Does https://github.com/ARMmbed/mbed-os/pull/6156 fix the issue you're seeing?

yossi2le, 2018-06-17:
@moranpeker 
I have tried to reproduce this bug but Its look like it is fixed. 
If this bug is no longer relevant please close it.

cmonr, 2018-06-18:
Going to close since it appears that the issue is resolved.
Please open another issue if this is not the case.

closed 2018-06-18 by cmonr