ARMmbed/mbed-os #5961: Support heap > 32KB for memory statistics

markus-becker-tridonic-com, 2018-01-29:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
ALL

**Toolchain:**
ALL

**Expected behavior**
Support heap > 32KB for memory statistics in 
https://github.com/ARMmbed/mbed-os/blob/729ef153076ea9db4a9c9f81e90076960e035213/features/FEATURE_COMMON_PAL/nanostack-libservice/mbed-client-libservice/nsdynmemLIB.h#L50

Change to int32_t or uint32_t for heap_sector_size, heap_sector_alloc_cnt, heap_sector_allocated_bytes, heap_sector_allocated_bytes_max.

markus-becker-tridonic-com, 2018-01-29:
FYI @daniel-cesarini-tridonic-com @MarceloSalazar

0xc0170, 2018-01-29:
@SeppoTakalo Please review

SeppoTakalo, 2018-01-29:
Tittle is a bit misleading as this only affects Nanostack's internal heap but still this is a valid issue.

kjbracey-arm, 2018-01-29:
The code linked is not the current version - that has a `ns_mem_heap_size_t` typedef.

Those 4 stats are limited by that size, as that's the type for size passed to ns_dyn_mem_init().

To up the heap liimit, you need to change that typedef. There's another typedef for the block size.

We do worry about 16-bit platforms a bit in that library, which is why they're 16-bit. I'd be inclined to maybe change them to `int`, so they stay 16-bit on those platforms and they become 32-bit on ARM.

markus-becker-tridonic-com, 2018-06-12:
@MarceloSalazar 32-bit nanostack heap is required on the K64 for larger wireless networks. At the moment the heap is still 16-bit:
https://github.com/ARMmbed/mbed-os/blob/master/features/frameworks/nanostack-libservice/mbed-client-libservice/nsdynmemLIB.h#L38

Could that be #ifdef'ed?

artokin, 2018-06-21:
Requested change is made in PR https://github.com/ARMmbed/nanostack-libservice/pull/68. It will be available in the next Mbed OS major release.

artokin, 2018-07-03:
ARMmbed/nanostack-libservice#68 is merged to Mbed OS master via PR https://github.com/ARMmbed/mbed-os/pull/7389.

closed 2018-07-03 by 0xc0170