ARMmbed/mbed-os #5976: Online compiler can't compile REALTEK RTL8195AM mbed-os-example-wifi

JanneKiiskila, 2018-01-30:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: [PR #5611](https://github.com/ARMmbed/mbed-os/pull/5611#issuecomment-361679737)
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
RTL8195AM

**Toolchain:**
on-line compiler

**Toolchain version:**
?

**mbed-cli version:** 
N/A

**mbed-os sha:**
mbed-OS 5.7.3

**DAPLink version:**
N/A

**Expected behavior**

Realtek RTL8195 AM compiles as per usual with the default mbed_app.json and target selected as RTL8195AM.

**Actual behavior**

It fails to compile, internal error.

""/extras/mbed_5e5da4a5990b/LPC1768/ARM/LPC1768.sct", line 4 (column 9): Error: L6235E: More than one section matches selector - cannot all be FIRST/LAST." in file "/"

Look for multiple mbed.bld files in your program.

**Steps to reproduce**

Open on-line compiler.
Select RTL8195AM as the target board
Import mbed-os-example-wifi
Compile

JanneKiiskila, 2018-02-01:
Internal ref: IOTTOO-3170

0xc0170, 2018-02-01:
Related https://github.com/ARMmbed/mbed-os/issues/5626 - there was a fix that was propagated to 5.7.3. Looks like the bug is still present.

I can reproduce this with blinky example and 5.7.4

@theotherjimmy

theotherjimmy, 2018-02-01:
Huh, I thought that #5626 would get the bug.

samchuarm, 2018-02-26:
Hi @0xc0170 @theotherjimmy any guidance we could share with Realtek for them to fix this issue would be helpful for Realtek Thanks.

theotherjimmy, 2018-02-26:
@samchuarm Thanks for the reminder. It looks like the post build script does not work correctly in the online compiler. I'll take a look.

theotherjimmy, 2018-02-26:
I fixed it on partnerbeta. I'm going to send a hotfix to the webt eam and a PR to Mbed OS in the next 20 min. Let's get this fixed.

theotherjimmy, 2018-02-26:
See https://github.com/ARMmbed/mbed-os/pull/6216

samchuarm, 2018-03-01:
Thanks so much @theotherjimmy

theotherjimmy, 2018-04-05:
This is not fixed by #6216

samchuarm, 2018-04-10:
@ARMmbed/team-realtek

samchuarm, 2018-04-10:
hi @theotherjimmy it looks like we're still seeing compile fails with online IDE on RTL8195AM target. Anything do you think we need Realtek team to address? Thanks.

samchuarm, 2018-04-10:
Btw, is it a new issue after merging #6216 ?

theotherjimmy, 2018-04-10:
I still need to figure out what's going wrong. It's the post-build script, but I don't know exactly what the problem is yet.

theotherjimmy, 2018-06-18:
@JanneKiiskila @samchuarm Good news: I can't reproduce this anymore! I think this is resolved. Please reopen with steps to reproduce if you find a project that is broken for only the Realtek board.

PR #6216 (2018-03-19): Correct Realtek post-build script to work in the online compiler
Resolves #5976
closed 2018-06-18 by theotherjimmy