ARMmbed/mbed-os #5981: "mbed export" always produces GettingStarted.html

drewcassidy, 2018-01-31:
### Description
- Type: Bug/Enhancement
- Priority: Minor

running `mbed export` to regenerate IDE files always produces a `GettingStarted.html` file in the project directory. Ideally this should only be generated the first time it is run, or removed entirely since it just links to the documentation for using the export tool (which the user must already know about to run the command...)

## Bug

**mbed-cli version:** 1.3.0

**mbed-os sha:** b87e98c57b976711c8f94920764eb1cd0631a6a4

**Steps to reproduce** run `mbed export` with any choice of IDE

0xc0170, 2018-01-31:
This was added here https://github.com/ARMmbed/mbed-os/pull/2671. It serves as a documentation pointer, it is part of exported projects from online IDE as well.

@theotherjimmy

theotherjimmy, 2018-01-31:
As martin said, this is by design. We could have it generate on the first time only, but I don't see an advantage to that (other than a very minor speedup).

drewcassidy, 2018-01-31:
I just feel the file is extraneous for me personally, especially since it regenerates after deleting it. maybe just a command line flag on `mbed export` to disable generating it?

vergil-SI, 2018-02-09:
A non regeneration flag would be much appreciated. This file haunts me. Every time I delete it then turn my head, there it is again.

theotherjimmy, 2018-02-09:
@vergil-SI Agreed on the haunting. I'm marking this as an enhancement. I have no idea what that flag would be. Let's use this issues to brainstorm names for this flag.

drewcassidy, 2018-02-09:
is there anything wrong with just having the `mbed export` tool in the command line just print out a URL instead of making a new file?

theotherjimmy, 2018-02-09:
@drewcassidy The `GettingStarted.htm` file is used by the online export system, which uses the same code.

theotherjimmy, 2018-02-09:
FYI, There is also a sneaky `.mbed` file being written to the root.

drewcassidy, 2018-02-10:
there could be a flag that the online tool uses to generate the file when needed

closed 2019-10-31 by ciarmcom