ARMmbed/mbed-os #5985: CLion exporter results in cmake build directory, causes compiler to fail in unexpected way

drewcassidy, 2018-01-31:
### Description
- Type: Bug
- Related issue: CMake exporter introduced with #5476 
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target:** any

**Toolchain:** `GCC_ARM`

**Toolchain version:** 7.2.1

**mbed-cli version:** 1.3.0

**mbed-os sha:** 195b3ea24b25c2711a11f5ee92570762105efb3c

**Expected behavior**
Exporter should automatically ignore `cmake-build-debug`, or at least remind the user to. Trying to compile with this directory generated by CLion should at least give a more clear error

**Actual behavior**
`mbed compile` fails with misleading errors when `cmake-build-debug` is present and generated by CLion

**Steps to reproduce**
- export CMakeLists.txt file using `mbed export -i cmake_gcc_arm`
- open the project in CLion and allow it to generate and fill the `cmake-build-debug` directory
- trying to compile a project that uses `ble/BLE.h` causes an unknown file or directory error, since the compiler tries to use source files in the cmake build directory (I'm unsure on *why* it does this)

**What should be done**
- the export tool should automatically add `cmake-build-debug` to `.mbedignore` (or place a wildcard ignore file in the build directory)
- If possible, the compiler should give a more useful error when trying to compile

0xc0170, 2018-02-01:
@thinkberg

thinkberg, 2018-02-02:
cab you assign it to me?

thinkberg, 2018-02-02:
@drewcassidy For clearification: you didn't compile using cmake/make after exporting, right? The issue arises using mbed compile directly or the cmake target mbed-cli-build. 

I will change the CMakeLists.txt to create an .mbedignore file in the binary output directory to ensure the cli build is not interfered with.

drewcassidy, 2018-02-02:
> @drewcassidy For clearification: you didn't compile using cmake/make after exporting, right? The issue arises using mbed compile directly or the cmake target mbed-cli-build.

yes, this is using `mbed compile`

PR #6001 (2018-02-07): Corrects intermittent CLion exporter issue with cmake build directory
## Description

adds a .mbedignore file to cmake binary dirs generated, which ensures they are ignored when compiling using the mbed-cli

## Status

**READY**

Fixes #5985
commit e321e98f (2018-02-07): fix #5985 CLion exporter results in cmake build directory, causes compiler to fail in unexpect...
closed 2018-02-07 by cmonr