ARMmbed/mbed-os #5989: some exporters clobber .mbedignore file

bmcdonnell-ionx, 2018-01-31:
### Description
- Type: Bug (arguably)
- Related issue: https://github.com/ARMmbed/mbed-cli/issues/378#issuecomment-362000601
- Priority: Minor

Some exporters (`e2studio`, `gnuarmeclipse`, and `mcuxpresso`, I think) overwrite the `.mbedignore` file, leaving no apparent way to generate an IDE project that has user-defined files excluded from the build.

Here's why I think it's just those three:

    /path/to/myproj/mbed-os/tools/export
    $ grep -R --binary-files=without-match mbedignore . | grep \.py
    ./e2studio/__init__.py:        self.gen_file('gnuarmeclipse/mbedignore.tmpl', jinja_ctx, '.mbedignore')
    ./gnuarmeclipse/__init__.py:        self.gen_file('gnuarmeclipse/mbedignore.tmpl', jinja_ctx, '.mbedignore')
    ./mcuxpresso/__init__.py:        self.gen_file('mcuxpresso/mbedignore.tmpl', jinja_ctx, '.mbedignore')

---------------------------------------------------------------
## Bug

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
96d9a00d0 (tag: mbed_lib_rev158, tag: mbed-os-5.7.3) Merge pull request #5841 from ARMmbed/release-candidate

**Expected behavior**
Adding files to `.mbedignore` causes them to be excluded from the build in any offline IDE for projects created by `mbed export`.

**Actual behavior**
Some exporters overwrite the `.mbedignore` file, so anything added to it by the user is overwritten, thus not ignored.

**Steps to reproduce**

    mbed new demoproj
    cd demoproj
    touch ignore-me-please.cpp
    
    echo ignore-me-please.cpp > .mbedignore
    echo .mbedignore before export:
    cat .mbedignore
    # try with -i {e2studio, gnuarmeclipse, mcuxpresso}
    mbed export -m LPC4088 -i mcuxpresso
    echo .mbedignore after export:
    cat .mbedignore

bmcdonnell-ionx, 2018-01-31:
Suggested behavior: if an exporter wants to ignore some files, it could parse `.mbedignore`, and append whatever it needs to add to it.

loverdeg-ep, 2018-01-31:
@janjongboom @screamerbg I believe this is one of the issues we were having with "mbed-cli" before I emailed you.

theotherjimmy, 2018-01-31:
@bmcdonnell-ionx I think I'll have the exporters do just that. I have a few other bugs queued for the exporters to look into, so I may not get to this bug this week.

loverdeg-ep, 2018-02-16:
@theotherjimmy 
We export gnuarmeclipse projects to get the benefits of the CDT indexer but still build with mbed-cli.

Any time we make a change to the project configuration we re-export the project; so this issue is very inconvenient.

theotherjimmy, 2018-02-16:
Ah, I'll look into this today. I think It should be simple to workaround.

PR #6121 (2018-02-20): Avoid blowing away .mbedignore files in exporters
Resolves #5989

@bmcdonnell-ionx, @loverdeg-ep Are there other files that could use the
same behavior?
closed 2018-02-20 by cmonr