ARMmbed/mbed-os #5991: How to enable multi-peripheral on nRF51822

drewcassidy, 2018-02-01:
### Description
- Type:  Question

---------------------------------------------------------------
## Question

I want to allow multiple peripherals to connect to my board, which should be supported by the S130 softdevice. I tried re-enabling advertising in the BLE connection callback, but when a second device attempts to connect it never completes.

drewcassidy, 2018-02-01:
Ah, I didnt realize that the default link count didnt allow this. I suspect this will be solved with #5980

pan-, 2018-02-01:
@drewcassidy What board are you using ? multiclient is disabled for 16K variants of the NRF51822 however it is enabled for 32K variants.

drewcassidy, 2018-02-01:
nRF51-DK, which has the 32K variant. Could this have changed with the unified nRF5 stuff?

desmond-blue, 2018-12-27:
Current Mbed-OS (the latest version is 5.11) is not able to support BLE multi-client on nRF51822.

nRF51822(or nRF51_DK) is using Nordic SDK 11 and SoftDevice S130 2.0.0, according to [the documentation](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s130.sds%2Fdita%2Fsoftdevices%2Fs130%2Fble_protocol_stack%2Frole_configuration.html&cp=3_7_2_0_9_3), there is a description.

> The SoftDevice provides a mechanism for enabling the number of Central or Peripheral roles the application can run concurrently. The SoftDevice can be configured with one connections as a Peripheral and up to eight connections as a Central. The SoftDevice supports running one Advertiser or Broadcaster and one Scanner or Observer concurrently with the BLE connections.

If enabling SoftDevice with peripheral count > 1, it returns NRF_ERROR_CONN_COUNT.

0xc0170, 2019-01-04:
cc @ARMmbed/mbed-os-pan

pan-, 2019-01-04:
@desmond-blue Acting as a GattClient is irrelative off the device role. A peripheral and a central can both act as GATT client and server concurrently. 

If the question is how to support multiple peripheral links, there is configuration options for it [here](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NORDIC_SOFTDEVICE/mbed_lib.json). Note that it may require to adjust `MBED_RAM_START` and `MBED_RAM_SIZE` to provision enough memory for the soft device otherwise the initialisation will fail (look [here](https://github.com/ARMmbed/mbed-os/blob/9ca15a77e197134efe0eff33667ad373378d4116/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_11/softdevice/common/softdevice_handler/softdevice_handler.c#L494-L507))

pan-, 2019-01-08:
According to the [softdevice specification](http://infocenter.nordicsemi.com/pdf/S130_SDS_v2.0.pdf) it is not possible to have more than one peripheral link: 

```
The SoftDevice provides a mechanism for enabling the number of Central or Peripheral roles the application can run concurrently. The SoftDevice can be configured with one connections as a Peripheral and up to eight connections as a Central. The SoftDevice supports running one Advertiser or Broadcaster and one Scanner or Observer concurrently with the BLE connections.
```

linlingao, 2019-05-30:
The last comment indicates it's not possible. The question has been answered. Closing.

closed 2019-05-30 by linlingao