ARMmbed/mbed-os #5992: RTC failing with latest master branch

akrnvm, 2018-02-01:
### Description
- Type: Bug 
- Related issue: `RTC`
- Priority: Blocker | Major 

---------------------------------------------------------------
## Bug

**Target**
K64F| Custom Board

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
RTC test case  " tests-mbed_drivers-rtc" will pass without errors
**Actual behavior**
Test Case is failing with latest mbedos (master branch)
**Steps to reproduce**
mbed test -m XXXXX -t ARM -v -n tests-mbed_drivers-rtc

----------------------------------------------------------------

## Question

How to fix this issue?
Why this extra sections are added in RTC tests?
Is this test case is PASSING with all other boards/controllers?

drewcassidy, 2018-02-01:
> Is this test case is PASSING with all other boards/controllers?

Test passes on NRF51-DK

akrnvm, 2018-02-01:
After RTC write with 1 sec wait delay, the test cases are passing ARM and GCC ARM but its fail in IAR.
What should be the reason its fail in new TEST case.
I have kept small delay after writing to stable RTC, is it cause this issue?

0xc0170, 2018-02-01:
latest master branch - please provide SHA

fails how - use `--v` - verbose output share where it fails and how

how to reproduce the error ? HAve you tried frdm k64f ? If not, how your custom board is different? This was tested on frdm k64f with all OK, we can run tests again and provide the output here

@mprse

akrnvm, 2018-02-01:
>latest master branch - please provide SHA

I have Cloned with HTTPS from master branch just now and tested.
Test same with mbedOS 5.7.4 and results are same.

My board is Cortex M3 controller.

Error log : 
```
[1517477275.41][HTST][INF] setting timeout to: 60 sec
[1517477275.45][SERI][INF] serial(port=COM6, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1517477275.45][SERI][INF] reset device using 'default' plugin...
[1517477275.70][SERI][INF] waiting 1.00 sec after reset
[1517477276.70][SERI][INF] wait for it...
[1517477276.70][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1517477276.70][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1517477276.70][CONN][INF] sending preamble 'c13554e9-4cb2-4281-aa61-a235819c3f7f'
[1517477276.70][SERI][TXD] {{__sync;c13554e9-4cb2-4281-aa61-a235819c3f7f}}
[1517477276.83][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1517477276.88][CONN][INF] found SYNC in stream: {{__sync;c13554e9-4cb2-4281-aa61-a235819c3f7f}} it is #0 sent, queued...
[1517477276.88][HTST][INF] sync KV found, uuid=c13554e9-4cb2-4281-aa61-a235819c3f7f, timestamp=1517477276.881000
[1517477276.90][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1517477276.90][HTST][INF] DUT greentea-client version: 1.3.0
[1517477276.92][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1517477276.93][HTST][INF] setting timeout to: 20 sec
[1517477276.96][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1517477276.97][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1517477276.97][HTST][INF] host test setup() call...
[1517477276.97][HTST][INF] CALLBACKs updated
[1517477276.97][HTST][INF] host test detected: default_auto
[1517477276.99][CONN][RXD] >>> Running 14 test cases...
[1517477277.05][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: attach stub RTC functions.}}, queued...
[1517477277.11][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: attach original RTC functions.}}, queued...
[1517477277.20][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - RTC functions are defined, RTC is enabled.}}, queued...
[1517477277.30][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - RTC functions are defined, RTC is disabled.}}, queued...
[1517477277.38][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - isenabled RTC function is undefined.}}, queued...
[1517477277.46][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - read RTC function is undefined.}}, queued...
[1517477277.53][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - result is stored in given buffer.}}, queued...
[1517477277.61][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: set_time() - RTC functions are defined.}}, queued...
[1517477277.68][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: set_time() - init RTC function is undefined.}}, queued...
[1517477277.77][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: set_time() - write RTC function is undefined.}}, queued...
[1517477277.84][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: set time - CUSTOM_TIME_0.}}, queued...
[1517477277.90][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: set time - CUSTOM_TIME_1.}}, queued...
[1517477277.97][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: set time - CUSTOM_TIME_2.}}, queued...
[1517477278.03][CONN][RXD]
[1517477278.03][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: RTC counts seconds.}}, queued...
[1517477278.10][CONN][RXD] >>> Running case #1: 'Unit Test: attach stub RTC functions.'...
[1517477278.16][CONN][INF] found KV pair in stream: {{__testcase_start;Unit Test: attach stub RTC functions.}}, queued...
[1517477283.84][CONN][RXD] :147::FAIL: Expected 1256729737 Was 1256729741
[1517477283.91][CONN][INF] found KV pair in stream: {{__testcase_finish;Unit Test: attach stub RTC functions.;0;1}}, queued...
[1517477284.02][CONN][RXD] >>> 'Unit Test: attach stub RTC functions.': 0 passed, 1 failed with reason 'Assertion Failed'
[1517477284.02][CONN][RXD]
[1517477284.08][CONN][RXD] >>> Test cases: 0 passed, 1 failed with reason 'Assertion Failed'
[1517477284.10][CONN][RXD] >>> TESTS FAILED!
[1517477284.13][CONN][INF] found KV pair in stream: {{__testcase_summary;0;1}}, queued...
[1517477284.15][CONN][INF] found KV pair in stream: {{max_heap_usage;0}}, queued...
[1517477284.15][HTST][ERR] orphan event in main phase: {{max_heap_usage;0}}, timestamp=1517477284.148000
[1517477284.17][CONN][INF] found KV pair in stream: {{reserved_heap;0}}, queued...
[1517477284.17][HTST][ERR] orphan event in main phase: {{reserved_heap;0}}, timestamp=1517477284.168000
[1517477284.19][CONN][INF] found KV pair in stream: {{end;failure}}, queued...
[1517477284.20][HTST][INF] __notify_complete(False)
[1517477284.20][HTST][INF] __exit_event_queue received
[1517477284.20][HTST][INF] test suite run finished after 7.27 sec...
[1517477284.21][CONN][INF] found KV pair in stream: {{__exit;0}}, queued...
[1517477284.21][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1517477284.22][HTST][INF] CONN exited with code: 0
[1517477284.22][HTST][INF] Some events in queue
[1517477284.22][HTST][INF] stopped consuming events
[... cut ...]
mbedgt: mbed-host-test-runner: stopped and returned 'OK'
mbedgt: test on hardware with target id: 701400000000000000000000000000000000000097969904
mbedgt: test suite 'tests-mbed_drivers-rtc' .......................................................... OK in 28.59 sec
        test case: 'RTC strftime' .................................................................... OK in 13.61 sec
mbedgt: test case summary: 1 pass, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.3151825783
mbedgt: test suite report:
```

mprse, 2018-02-01:
> :147::FAIL: Expected 1256729737 Was 1256729741

It looks like RTC does not count seconds. From the results 4 seconds elapsed between successive calls `set_time(CUSTOM_TIME_1);` and `seconds = time(NULL);`. We expect that same value should be retrieved.

Cortex M3 is used in NCS36510 and there was the same problem. Please check issue https://github.com/ARMmbed/mbed-os/issues/5308. Unfortunately fix for this issue is not available, you can check RTC driver or disable RTC (`targets.json` file) support for now.

akrnvm, 2018-02-02:
Hi,
I have added one second delay after rtc write to stable.
with delay the 13 test cases are passing.

```
[1517564720.84][HTST][INF] starting host test process...
[1517564721.84][CONN][INF] starting connection process...
[1517564721.84][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1517564721.84][CONN][INF] initializing serial port listener...
[1517564721.84][PLGN][INF] Waiting up to 60 sec for 'XXXX00000000000000000000000000000000000097969904' serial port (current is 'COM6')...
[1517564721.85][HTST][INF] setting timeout to: 60 sec
[1517564721.90][SERI][INF] serial(port=COM6, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1517564721.90][SERI][INF] reset device using 'default' plugin...
[1517564722.15][SERI][INF] waiting 1.00 sec after reset
[1517564723.15][SERI][INF] wait for it...
[1517564723.15][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1517564723.15][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1517564723.15][CONN][INF] sending preamble 'e2db5a57-48d2-45d8-b248-173383bb157b'
[1517564723.15][SERI][TXD] {{__sync;e2db5a57-48d2-45d8-b248-173383bb157b}}
[1517564723.29][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1517564723.34][CONN][INF] found SYNC in stream: {{__sync;e2db5a57-48d2-45d8-b248-173383bb157b}} it is #0 sent, queued...
[1517564723.34][HTST][INF] sync KV found, uuid=e2db5a57-48d2-45d8-b248-173383bb157b, timestamp=1517564723.338000
[1517564723.36][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1517564723.36][HTST][INF] DUT greentea-client version: 1.3.0
[1517564723.38][CONN][INF] found KV pair in stream: {{__timeout;20}}, queued...
[1517564723.38][HTST][INF] setting timeout to: 20 sec
[1517564723.42][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1517564723.42][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1517564723.43][HTST][INF] host test setup() call...
[1517564723.43][HTST][INF] CALLBACKs updated
[1517564723.43][HTST][INF] host test detected: default_auto
[1517564723.44][CONN][RXD] >>> Running 14 test cases...
[1517564723.50][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: attach stub RTC functions.}}, queued...
[1517564723.58][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: attach original RTC functions.}}, queued...
[1517564723.66][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - RTC functions are defined, RTC is enabled.}}, queued...
[1517564723.75][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - RTC functions are defined, RTC is disabled.}}, queued...
[1517564723.83][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - isenabled RTC function is undefined.}}, queued...
[1517564723.91][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - read RTC function is undefined.}}, queued...
[1517564723.98][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: time() - result is stored in given buffer.}}, queued...
[1517564724.06][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: set_time() - RTC functions are defined.}}, queued...
[1517564724.14][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: set_time() - init RTC function is undefined.}}, queued...
[1517564724.22][CONN][INF] found KV pair in stream: {{__testcase_name;Unit Test: set_time() - write RTC function is undefined.}}, queued...
[1517564724.29][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: set time - CUSTOM_TIME_0.}}, queued...
[1517564724.35][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: set time - CUSTOM_TIME_1.}}, queued...
[1517564724.42][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: set time - CUSTOM_TIME_2.}}, queued...
[1517564724.49][CONN][RXD]
[1517564724.49][CONN][INF] found KV pair in stream: {{__testcase_name;Functional Test: RTC counts seconds.}}, queued...
[1517564724.55][CONN][RXD] >>> Running case #1: 'Unit Test: attach stub RTC functions.'...
[1517564724.61][CONN][INF] found KV pair in stream: {{__testcase_start;Unit Test: attach stub RTC functions.}}, queued...
[1517564725.76][CONN][INF] found KV pair in stream: {{__testcase_finish;Unit Test: attach stub RTC functions.;1;0}}, queued...
[1517564725.84][CONN][RXD] >>> 'Unit Test: attach stub RTC functions.': 1 passed, 0 failed
[1517564725.84][CONN][RXD]
[1517564725.90][CONN][RXD] >>> Running case #2: 'Unit Test: attach original RTC functions.'...
[1517564725.97][CONN][INF] found KV pair in stream: {{__testcase_start;Unit Test: attach original RTC functions.}}, queued...
[1517564727.04][CONN][INF] found KV pair in stream: {{__testcase_finish;Unit Test: attach original RTC functions.;1;0}}, queued...
[1517564727.12][CONN][RXD] >>> 'Unit Test: attach original RTC functions.': 1 passed, 0 failed
[1517564727.12][CONN][RXD]
[1517564727.20][CONN][RXD] >>> Running case #3: 'Unit Test: time() - RTC functions are defined, RTC is enabled.'...
[1517564727.29][CONN][INF] found KV pair in stream: {{__testcase_start;Unit Test: time() - RTC functions are defined, RTC is enabled.}}, queued...
[1517564727.39][CONN][INF] found KV pair in stream: {{__testcase_finish;Unit Test: time() - RTC functions are defined, RTC is enabled.;1;0}}, queued...
[1517564727.48][CONN][RXD] >>> 'Unit Test: time() - RTC functions are defined, RTC is enabled.': 1 passed, 0 failed
[... cut ...]
        test case: 'Unit Test: time() - isenabled RTC function is undefined.' ........................ OK in 0.09 sec
        test case: 'Unit Test: time() - read RTC function is undefined.' ............................. OK in 0.08 sec
        test case: 'Unit Test: time() - result is stored in given buffer.' ........................... OK in 0.08 sec
mbedgt: test case summary: 14 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.1341709337
mbedgt: test suite report:
```
Is it ok to change the test time?
How can  I adjust my RTC stable time with TEST cases?

mprse, 2018-02-02:
> I have modified the test case timeout from 20 to 22

This can be easily modified in the test.

> I have added one second delay after rtc write to stable.

Can you provide more details about this fix? Where did you added this delay? Please provide a piece of code with changes.

akrnvm, 2018-02-02:
>Can you provide more details about this fix? Where did you added this delay? Please provide a piece of code with changes.
```

void rtc_write(time_t t)
{
    struct tm *timeinfo = localtime(&t);  // Call local time API to get time zone
    diff_year = timeinfo->tm_year - (timeinfo->tm_year % 100);
     <snip>    
     .....
     ......
    <snip>
    XXX_RTC->PAGER &= PAGER_PAGE_ZERO;  // Select PAGE 0
    XXX_RTC->RESTR |= RTC_RESET;
    XXX_RTC->PAGER &= RTC_CLK_DISABLE;  // Disable clock

    XXX_RTC->YEARR = (uint8_t)DEC2HEX((timeinfo->tm_year - diff_year));  // Set year value
    XXX_RTC->MONTHR = (uint8_t)DEC2HEX(timeinfo->tm_mon);  // Set month value
    XXX_RTC->DATER = (uint8_t)DEC2HEX(timeinfo->tm_mday);  // Set date value
    XXX_RTC->DAYR = (uint8_t)(timeinfo->tm_wday);  // Set week day value
    XXX_RTC->HOURR = (uint8_t)DEC2HEX(timeinfo->tm_hour);  // Set hour value
    XXX_RTC->MINR = (uint8_t)DEC2HEX(timeinfo->tm_min);  // Set minute value
    XXX_RTC->SECR = (uint8_t)DEC2HEX(timeinfo->tm_sec);  // Set second value

    XXX_RTC->PAGER |= RTC_CLK_ENABLE;  // Enable Clock
    wait_ms(1000); // time to stable RTC 
}
```

mprse, 2018-02-02:
I understand that this is your own RTC write driver. It looks like additional time is needed probably since RTC operates on different clock than CPU. Please check the documentation since 1 sec delay is not a valid solution. On `NCS36510` board which is also based on Cortex-M3 the following solution is provided:
https://github.com/ARMmbed/mbed-os/blob/b8d218038bcd8c2ad447d91ab94c927118c1ad61/targets/TARGET_ONSEMI/TARGET_NCS36510/rtc.c#L260

Note that this board has a similar problem.

akrnvm, 2018-02-05:
Hi,
Thank you for sharing the reference.

I have tried with RTC interrupt method(1Hz) and able to execute test case without time_ms(1000).
But total time is exceeding  20 seconds.
Test case is passing with 22 seconds:
```
mbedgt: test suite 'tests-mbed_drivers-rtc' .......................................................... OK in 35.54 sec
        test case: 'Functional Test: RTC counts seconds.' ............................................ OK in 11.05 sec
        test case: 'Functional Test: set time - CUSTOM_TIME_0.' ...................................... OK in 1.06 sec
        test case: 'Functional Test: set time - CUSTOM_TIME_1.' ...................................... OK in 1.07 sec
        test case: 'Functional Test: set time - CUSTOM_TIME_2.' ...................................... OK in 1.08 sec
        test case: 'Unit Test: attach original RTC functions.' ....................................... OK in 1.08 sec
        test case: 'Unit Test: attach stub RTC functions.' ........................................... OK in 1.06 sec
        test case: 'Unit Test: set_time() - RTC functions are defined.' .............................. OK in 0.08 sec
        test case: 'Unit Test: set_time() - init RTC function is undefined.' ......................... OK in 0.09 sec
        test case: 'Unit Test: set_time() - write RTC function is undefined.' ........................ OK in 0.09 sec
        test case: 'Unit Test: time() - RTC functions are defined, RTC is disabled.' ................. OK in 0.10 sec
        test case: 'Unit Test: time() - RTC functions are defined, RTC is enabled.' .................. OK in 0.09 sec
        test case: 'Unit Test: time() - isenabled RTC function is undefined.' ........................ OK in 0.08 sec
        test case: 'Unit Test: time() - read RTC function is undefined.' ............................. OK in 0.08 sec
        test case: 'Unit Test: time() - result is stored in given buffer.' ........................... OK in 0.08 sec
mbedgt: test case summary: 14 passes, 0 failures

```
My controller has to wait 1Hz interrupt for proper writing  and reading 

[image]

This will causing more than 1 sec delay in "test case: 'Functional Test: RTC counts seconds.'"
 
Any solution for this issue?

0xc0170, 2018-02-05:
As I understand, the problem here is the test timeout. For your target, 22 seconds is needed to execute the test.

I wonder about how long this test takes `        test case: 'Functional Test: RTC counts seconds.' ............................................ OK in 11.05 sec`. The timeout increase would be fine as long as it has a valid reasons to do (did we set the test timeout too low? How long it takes for other targets? Is there a big difference in the time execution - if yes, then why this target needs 22 seconds).

Why is that 1second delay needed, there is no flag set when RTC is ready? That would be more deterministic.

akrnvm, 2018-02-06:
>I wonder about how long this test takes test case: 'Functional Test: RTC counts seconds.' ............................................ OK in 11.05 sec. 

This test case is writing data to RTC and waiting 10 sec and its reading again RTC.
writing time(~1sec) + wait(10 sec) + readtime(~1sec) = 11.05 sec

>I don't fully understand 1HZ interrupt (is that flow present in rtc read/write functions)?

As per datasheet this is the method to write and read RTC.
The 1Hz interrupt is generated by being synchronized with counting up of the second counter. If
data is written in the time between 1Hz interrupt and subsequent one second count, it completes correctly.
Data can be read correctly if reading data after 1Hz interrupt occurred.

>Why is that 1 second delay needed, there is no flag set when RTC is ready? That would be more deterministic

No,there is no extra flags or bits are available to see RTC status. Only writing accept between 1Hz interrupt and subsequent one second count.

donatieng, 2018-06-20:
@akrnvm are you still experiencing the issue?

closed 2018-12-19 by ciarmcom