ARMmbed/mbed-os #5999: SD card FAT format with multipartitions are not compatible with PC

JaniSuonpera, 2018-02-02:
### Description
- Type: Bug 
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**mbed-os sha:**
(`git log -n1 --oneline`)
caeaa49 (HEAD, tag: mbed-os-5.7.4, origin/mbed-os-5.7) Merge pull request #5954 from ARMmbed/release-candidate

**Expected behavior**
- SD contains two partition which can not be read with PC but target K64F is still able to read/write SD card without any problem. 

**Actual behavior**
- SD parts partition type is unknown. (check with PC e.g. Linux program  gparted )

**Steps to reproduce**
Initialize BlockDevice point to SD e.g. with
   BlockDevice *bd = storage_selector();

   // Create a partitions
   MBRBlockDevice::partition(bd, 1, 0x83, 0, 1024*1024);
   MBRBlockDevice::partition(bd, 2, 0x83, 1024*1024, 2*1024*1024);

   // Create the block device that represents the partition
   MBRBlockDevice part1(bd, 1);
   MBRBlockDevice part2(bd, 2);

   // Format the partitions with a FAT filesystem
   FATFileSystem::format(&part1);
   FATFileSystem::format(&part2);

Check SD card with PC.

cmonr, 2018-02-02:
@JaniSuonpera, are you attempting to access this partitioned micro SD card in Windows? Afaik, the Windows driver does not support viewing partitioned removable media.

If you're attempting to view this in Linux, could you provide system information on the machine?

@geky

JaniSuonpera, 2018-02-05:
@geky I have Lenovo desktop Linux (Ubuntu 16.04 LTS) PC. I have tested this with external (https://www.transcend-info.com/Products/No-331) and internal SD card reader. Result is same after formatting with K64F.

If I create partitions (FAT32) with Linux program gparted then Linux filemanager is able to mount both partitions from SD card. Same is case also with Windows 10 (Lenovo T460s) laptop with external SD card reader (https://www.transcend-info.com/Products/No-331). Formatting available partitions from SD card with mbed-os makes SD unreadable with PC. Linux shows that partition type is unknown.

There seems to be some issue to use MBRBlockDevice. If initialize CD card with PC to FAT32 single partition mode and use only BlockDevice pointer for format then SD card is readable with PC. I use MBRBlockDevice pointer for format then SD card is unreadable with PC.

FYI @JanneKiiskila

geky, 2018-02-13:
Hi @JaniSuonpera, sorry for the delay.

It looks like there was a regression in the FAT filesystem regarding extra MBR blocks. Does #6074 fix your issue?

JaniSuonpera, 2018-02-13:
Hi @geky, I tested  #6074. Formatting single partition FAT32 SD card is working fine.

For dual partition it depends, which kind partitions exists in SD card. If there are two FAT32 primary partitions then after format there are two FAT16 primary partitions. If there is one FAT32 primary partition and extended partition, which contains one FAT32 logical partition, then after format partition one is FAT16 and partition two is unknown type.

If create partitions with  MBRBlockDevice::partition and format then both will be FAT16 primary partitions.

How I can create FAT32 partitions with MBRBlockDevice::partition? How about logical partitions? It is logical partitions supported?

geky, 2018-02-13:
Logical partitions are currently not supported. Though it would be a good idea to error rather than allowing the filesystem to overwrite logical partitions. I added a check for logical partitions to https://github.com/ARMmbed/mbed-os/pull/6074.

---

As for why you can't create a FAT32 partition: it's not allowed. After partitioning your block device the number of clusters are small enough that they fit in FAT16, so it **must** be FAT16. There's actually no other identifier that marks the difference between the filesystems other than the cluster count.

Try formatting the FAT filesystem on your PC, it should still show up as FAT16.

Fortunately, FAT12/FAT16/FAT32 are all handled the same by host OSs. So you should be able to mount and use the filesystem without worrying about the difference between FAT12/FAT16/FAT32.

I wouldn't think of these as three different filesystems, because in order to support the full range of block device sizes you need to support all three.

JaniSuonpera, 2018-02-14:
I approve #6074. Please release it next mbed-os 5.7.X release.

These limitations must document somewhere.

closed 2018-02-14 by cmonr

geky, 2018-02-14:
Noted, I made another pr to mention this: https://github.com/ARMmbed/mbed-os/pull/6095. 

This patch has already been merged, so you can expect it on 5.7.6.

JanneKiiskila, 2018-02-14:
Thanks @geky  for good & quick support!

geky, 2018-02-14:
Also thank @cmonr and the maintainers for the quick turnaround on prs : )

cmonr, 2018-02-15:
@geky It's easy when the PRs are small and well documented :)