ARMmbed/mbed-os #6002: TARGET_NRF52_COMMON and PWM

berrg, 2018-02-02:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement

---------------------------------------------------------------
## Bug

**Target**
TARGET_NRF52_COMMON

**Toolchain:**
ALL

**Expected behavior**
12 channel PWM
**Actual behavior**
3 channel bahavior

**Steps to reproduce**
You can only assign three PWM channels since the code only use one channel per PWM module.
----------------------------------------------------------------
## Enhancement

**Suggested enhancement**

Modify  mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_NRF52_COMMON/pwmout_api.c
to use all channels. The NRF chip has three PWM modules with 4 channels each, total 12 channels.

0xc0170, 2018-02-07:
@ARMmbed/team-nordic @pan-  Please review

marcuschangarm, 2018-03-08:
Unfortunately, this is not possible on the current hardware and API since the PWM period is shared across all channels on each PWM module, so changing the period on one PWM object would affect the period on another PWM object.

0xc0170, 2018-05-18:
> Unfortunately, this is not possible on the current hardware and API since the PWM period is shared across all channels on each PWM module, so changing the period on one PWM object would affect the period on another PWM object.

@berrg This should be resolved as won't fix

closed 2018-05-18 by 0xc0170