ARMmbed/mbed-os #6008: CI tests failing for filesystem tests on targets with 32K RAM

ashok-rao, 2018-02-04:
### Description
- Type: Enhancement | Question
- Related PR: `#5975`
- Priority: Major

## Enhancement

**Some of the targets with 32K RAM seems to be failing the heap block device tests**

**Suggested enhancement**
Can the tests be modified to use less heap on resource constrained devices?

**Pros**
Useful for many smaller targets with limited RAM.
-----------------------------------------------------------------

## Question

**How to?**
Is there an option to include something like "special, low memory" tests that could may be selected based on a macro (similar to that being done for SD driver tests now) and built / ran only for a few selected targets?

ashok-rao, 2018-02-04:
@geky @theotherjimmy @cmonr @0xc0170 .. this issue is for tracking the specific test changes made for xDOT as part of https://github.com/ARMmbed/mbed-os/pull/5975 ..

cmonr, 2018-07-18:
>  Is there an option to include something like "special, low memory" tests that could may be selected based on a macro (similar to that being done for SD driver tests now) and built / ran only for a few selected targets?

This sounds like something that @ARMmbed/mbed-os-storage just brought in.

davidsaada, 2018-07-18:
> This sounds like something that @ARMmbed/mbed-os-storage just brought in.

Indeed. [#7465](https://github.com/ARMmbed/mbed-os/pull/7465) handles it.

closed 2019-06-05 by ciarmcom