ARMmbed/mbed-os #6010: Cortex-M7 cache setting

bcostm, 2018-02-05:
### Description
- Type: Question
- Targets: Cortex-M7
- Priority: Major?

-----------------------------------------------------------------

## Question

As of now, the cache is enabled in targets/TARGET_STM/**mbed_overrides.c** (example for STM32F7 devices). **But shouldn't this be generic** ?

Also in case of **bootloader**, shoudn't mbed_start_application handle the cache cleaning / disabling before jumping into new application ?

This question is a follow-up of a comment posted in PRs #5972 while implementing the bootloader on STM32F7 devices.

cc @ARMmbed/team-st-mcd

0xc0170, 2018-02-05:
@bulislaw @c1728p9 @SenRamakri  Please review

bulislaw, 2018-02-07:
I agree it should probably happen somewhere in mbed_boot.c, to be fair this file is a mess and in need of refactoring anyway. We've managed to keep it almost (M0 and VTOR exception) core agnostic and I feel a bit bad to add yet another `#ifdef` there, but it makes sense.

@c1728p9 what do you think about bootloader case?

c1728p9, 2018-02-08:
> Also in case of bootloader, shoudn't mbed_start_application handle the cache cleaning / disabling before jumping into new application ?

The should be handled in the application's boot sequence regardless, to ensure it works with different bootloader variants.

SenRamakri, 2018-02-09:
Enabling of Icache/Dcache can be done dynamically(without ifdefs) by detecting Arch version in CPUID and presence of Cache support in CLIDR. But we may not be able to CMSIS structs( SCB ). I agree that we should do this during Mbed boot and not in bootloader as @c1728p9 mentioned to keep the bootloader generic/simple.

cmonr, 2018-06-18:
@studavekar Fyi, since I remember we had really strange CI issues when M7 cache was enabled at first.

alfred-ai, 2018-07-25:
We have seen mysterious hardfaulting and MCU stalling when trying enable I/D-cache when it is already enabled (vice-versa for disabling cache when it is already disabled) on Atmel's Cortex-M7 (SAME/S/V70 families).

Our solution is to first check whether it is enable/disabled before enable/disabling it, e.g.:
```c
void SCB_EnableDCache (void)
{
...
if(SCB->CCR & (uint32_t)SCB_CCR_DC_Msk)
{
   // D-cache already enabled, return 
   // (reenabling it will halt the system)
   return;
```

closed 2019-10-31 by ciarmcom