ARMmbed/mbed-os #6038: Upgrade memap to show components of included libraries

AGlass0fMilk, 2018-02-07:
### Description
- Type: Enhancement 
- Priority: Minor

----------------------------------------------------------------
## Enhancement
When statically compiling mbed libraries with mbed-cli using the --library flag it appears that the memap tool doesn't know how to output the various sections. I'm not sure if this is a limitation of memap or just due to the nature of statically linked libraries.

My build process statically builds a bunch of libraries together, including mbed-os and some in-house libraries. When memap displays post-build analysis, these statically linked libraries appear as "../.." which isn't very helpful in discovering where all your flash consumption is coming from.

Example:
```
+-------------------------+-------+-------+------+
| Module                  | .text | .data | .bss |
+-------------------------+-------+-------+------+
| ../..                   | 24943 |    72 | 1304 |
| BUILD/image_info.o      |   512 |     0 |    0 |
| BUILD/irqn_workaround.o |    78 |     0 |    0 |
| BUILD/main.o            |  1548 |    24 |  206 |
| [fill]                  |   335 |     4 |   26 |
| [lib]/c.a               |  2800 |  2108 |   56 |
| [lib]/gcc.a             |   808 |     0 |    0 |
| [lib]/misc              |   252 |    12 |   28 |
| Subtotals               | 31276 |  2220 | 1620 |
+-------------------------+-------+-------+------+
Total Static RAM memory (data + bss): 3840 bytes
Total Flash memory (text + data): 33496 bytes
```

**Reason to enhance or problem with existing solution**
Upgrade memap to show components of included libraries so size optimizations can be carried out more easily. This would allow one to see what statically linked library components are taking up unnecessary flash space.

0xc0170, 2018-02-08:
@theotherjimmy Pleas review

theotherjimmy, 2018-02-08:
@AGlass0fMilk Do you include them with `--source`?

AGlass0fMilk, 2018-02-08:
I am including them with the --source flag, is there another way to include static libraries?

theotherjimmy, 2018-02-13:
@AGlass0fMilk You could copy, or symlink the library directory into the project directory as well. That method should get you something other than `../..` as the module name. Further, you can pass `--stats-depth` with numbers larger than 2 to `mbed compile` to show more detail

----

As for this enhancement, I'm going to have think on how to do that for a bit. I welcome any suggestions.

AGlass0fMilk, 2018-02-13:
I created new build tools that give the fully qualified paths to my other libraries to the mbed compile command. I also found the --stats-depth command allowed me to see much more output (and I also looked at the newish flash memory visualizer tool -- very cool!)

One question I do have: Why isn't the --stats-depth command documented in the mbed compile -h screen? There seems to be a few undocumented commands that would be useful for more advanced use cases.

theotherjimmy, 2018-02-14:
@AGlass0fMilk It's not documented because mbed-cli maintains it's own, simplified help. I think a feature to show the "advanced" help could be useful in this case. You can run `python mbed-os/tools/make.py -h` to see the full list of option supported by `mbed compile` for now.

theotherjimmy, 2018-06-19:
@AGlass0fMilk My recent PR, #7183, actually resolves the `../` issues with `--source` arguments. I think that's all that is remaining on this issue, correct?

AGlass0fMilk, 2018-06-19:
I believe so, closing this issue. 👍

PR #7183 (2018-07-17): Tools: Extract resources object and implement incremental scan
### Description

I found that the resource object being part of the tool chain, and depending on 
tool chain infrastructure was a bit odd, so I extracted resources from tool 
chains. The resulting code is quite a bit more clear in it's purpose and can be a 
tad smaller (extra documentation and formatting excluded). I made the resources 
object fully incremental, meaning that the scan stops early otfen for targets, 
toolchains and features, and can easily pick itself back up where it left off.

A side effect of fully incremental sca [... cut]
closed 2018-06-19 by AGlass0fMilk