ARMmbed/mbed-os #6039: ADC for Nordic devices

berrg, 2018-02-08:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#abc`
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF52-dk (Possible others Nordic targets)

**Expected behavior**
From Mbed documentation:
Use the AnalogIn API to read an external voltage applied to an analog input pin. **AnalogIn() reads the voltage as a fraction of the system voltage.** The value is a floating point from 0.0(VSS) to 1.0(VCC). For example, if you have a 3.3V system and the applied voltage is 1.65V, then AnalogIn() reads 0.5 as the value.

**Actual behavior**
NRF52-dk ADC configuration in nrf_drv_saadc.h use internal referance voltage and not VCC.

**Suggested enhancement**
Change these parameter in nrf_drv_saadc.h (for both for SE and DIFFERENTIAL)

```
    .gain       = NRF_SAADC_GAIN1_6,                 
    .reference  = NRF_SAADC_REFERENCE_INTERNAL,       
```
To the following:
```
  .gain       = NRF_SAADC_GAIN1_4,                  
  .reference  =  NRF_SAADC_REFERENCE_VDD4,       
```

0xc0170, 2018-02-08:
@ARMmbed/team-nordic Please review

marcuschangarm, 2018-03-08:
@berrg 

We are in the process of updating the NRF52 SDK to version 14.2 and the new ADC driver is using the external reference voltage as suggested: https://github.com/ARMmbed/mbed-os/pull/6223

You can follow the progress here:
https://os.mbed.com/forum/upcoming-features/topic/29477/

Please give it a whirl and see if this solves your problem! Thanks!

tkstreet, 2018-05-02:
> Please give it a whirl and see if this solves your problem! Thanks!

Give *what* a whirl?  There's no suggestions/instruction on either of those pages?

tkstreet, 2018-05-03:
Nope.  It's still not in the master.  And I can't pull the feature branch into mbed.  I even tried forking the master branch and making the edits myself and pulling that into mbed, but now it is only producing .bin files instead of .hex files and it has screwed up the BLE GATT server.  I don't even understand this, because they should be identical otherwise.

marcuschangarm, 2018-05-04:
```
    nrf_saadc_channel_config_t channel_config = {
        .resistor_p = NRF_SAADC_RESISTOR_DISABLED,
        .resistor_n = NRF_SAADC_RESISTOR_DISABLED,
        .gain       = NRF_SAADC_GAIN1_4,
        .reference  = NRF_SAADC_REFERENCE_VDD4,
        .acq_time   = NRF_SAADC_ACQTIME_10US,
        .mode       = NRF_SAADC_MODE_SINGLE_ENDED,
        .burst      = NRF_SAADC_BURST_DISABLED,
        .pin_p      = input,
        .pin_n      = NRF_SAADC_INPUT_DISABLED
    };
```

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/analogin_api.c#L82-L95

tkstreet, 2018-05-04:
I don't get it.  All I can tell you is that my ADC is still reading a fraction of what it should be with the master branch updated, and it's perfectly fine when I modify nrf_drv_saadc.h, fork it and use that instead, with no other changes whatsoever, but that it screws up the BLE Gatt server and whatnot.

marcuschangarm, 2018-05-04:
I can't reproduce what you are describing.

```
#include "mbed.h"

AnalogIn adc0(A0);

int main(void)
{
    float value0 = adc0.read();

    printf("adc: %f\r\n", value0);
}
```

This prints out `0.0`, `1.0`, and `0.5` when I hook up `GND`, `VDD`, and a voltage divider to `A0` on NRF52_DK/GCC_ARM.

marcuschangarm, 2018-05-10:
Ah, you are using the online compiler! I don't think it is possible to checkout the `master` branch directly, it only allows you to checkout tagged releases. 

Unfortunately, you'll have to wait until 5.9 is released then or alternatively try out the command line build tools.

tkstreet, 2018-05-11:
I was afraid you were going to say that. Haha. It seems like a very precarious situation to have a setup where the same library pulls two different commits and compiles potentially completely different functionality (as is the case here) depending on _what_ environment you are performing the operations in.

Anyway, thanks for the help. I think I'm just going to have to phase out the online compiler entirely from our development to avoid this situation in the future.

0xc0170, 2018-06-19:
@tkstreet @berrg @marcuschangarm Is this issue not fixed? 5.9 was released

0xc0170, 2018-10-04:
This can be closed as 5.9 should have addressed this issue.

closed 2018-10-23 by ciarmcom