ARMmbed/mbed-os #6060: gnuarmeclipse exporter ignores .mbedignore files

janjongboom, 2018-02-12:
Maybe related to #5989?

### Description
- Type: Bug
- Related issue: #5989

---------------------------------------------------------------
## Bug

When exporting to gnuarmeclipse mbedignore files are ignored, and all files are added to the new project. This breaks compilation for projects that have other `main` files, but that are ignored by mbedignore. Normal building with Mbed CLI does not yield any issues.

See https://github.com/janjongboom/janpatch/issues/7

theotherjimmy, 2018-02-12:
@janjongboom That's simply not true. Do you get a "tools" directory in your project? I don't.

theotherjimmy, 2018-02-12:
Also, while #5989 seems related, it's not. The exporters resolve the `.mbedignore` files before writing anything to disk.

amq, 2018-02-12:
@theotherjimmy do you have any idea why https://github.com/janjongboom/janpatch/blob/master/.mbedignore doesn't exclude `janpatch-cli.c`?

theotherjimmy, 2018-02-12:
I think the exclude rules in the exporter are not setup for single file excludes.

janjongboom, 2018-02-13:
@theotherjimmy interesting, as the mbed CLI build process properly excludes them.

theotherjimmy, 2018-02-20:
@janjongboom I forgot a line in the original implementation see #6145 For the correction!

janjongboom, 2018-02-21:
Great, thanks.

PR #6145 (2018-02-22): Correct single file excludes for some exporters
Resolves #6060
closed 2018-02-21 by janjongboom