ARMmbed/mbed-os #6062: OdinWiFiInterface uses too small buffer to contain IPv6 address

juhaylinen, 2018-02-12:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
04f0f2b1a

**Expected behavior**
"get_ip_address()” call in OdinWiFiInterface returns IPv6 address when IPv6 support is enabled and connected to WiFi AP

**Actual behavior**
"get_ip_address()” call in OdinWiFiInterface returns NULL address. The reason seems to be too small a buffer to contain an IPv6 address.

**Steps to reproduce**
Requires IPv6 capable Wifi router
Configure mbed-os-example-wifi to use IPv6 only, compile and run.

0xc0170, 2018-02-12:
@ARMmbed/team-ublox Please review

juhaylinen, 2018-06-13:
This issue seems to be fixed after EMAC update

closed 2018-06-13 by juhaylinen