ARMmbed/mbed-os #6065: [NRF52] FlashIAP Interface usage without SoftDevice

drahnr, 2018-02-12:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug/Question
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NRF52_DK / Custom

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (Fedora 7.1.0-5.fc27) 7.1.0

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
4.7.4

**Expected behavior**
FlashIAP works 

**Actual behavior**
FlasIAP programing fails with ret code -1

**Steps to reproduce**
custom_targets.json
```
{
    "CUSTOM_TARGET": {
        "supported_form_factors": ["ARDUINO"],
        "inherits": ["NRF52_DK"],
        "macros_remove": ["S132", "SOFTDEVICE_PRESENT", "CONFIG_GPIO_AS_PINRESET", "BLE_STACK_SUPPORT_REQD"],
        "macros_add" : ["CONFIG_NFCT_PINS_AS_GPIOS","MBED_TICKLESS"],
        "release_versions": ["2", "5"],
        "device_has_add": ["FLASH"],
        "device_name": "nRF52832_xxAA",
        "features_remove": ["BLE"],
        "MERGE_BOOTLOADER": false,
        "MERGE_SOFTDEVICE": false,
        "bootloader_supported": true
    }
}
```

```
#if 1
    ret = iap.init(); // <<<< good
    if (ret < 0 ) {
        printf("Failed to init IAP %i", ret);
    } else {
        const char x[] = "TestPatternABC1230xFF";
        const size_t x_len = sizeof(x);
        char y[x_len] = {'\0', };
        const uint32_t addr = 0x38000; // somewhere after the firmware
        ret = iap.program(x, addr, x_len); // <<<<<< fails
        if (ret < 0) {
            printf("Failed to program IAP %i", ret);
        } else {
            ret = iap.read(y, addr, x_len);
            if (ret < 0) {
                printf("Failed to read IAP %i", ret);
            } else {
                assert(memcmp(y,x,x_len) == 0);
            }
            printf("Read %s", y);
        }
        ret = iap.deinit();
        if (ret < 0) {
            printf("Failed to deinit IAP %i", ret);
        }
    }
#endif
```
----------------------------------------------------------------

## Question

**How to?**

So apparently the check of `sd_softdevice` in  

```
int32_t flash_program_page(flash_t *obj, uint32_t address, const uint8_t *data, uint32_t size)
{
    uint8_t sd_enabled;
    if ((sd_softdevice_is_enabled(&sd_enabled) == NRF_SUCCESS) && sd_enabled == 1) {
        return -1;
    }
    /* We will use *_words function to speed up flashing code. Word means 32bit -> 4B
     * or sizeof(uint32_t).
     */
    nrf_nvmc_write_words(address, (const uint32_t *) data, (size / sizeof(uint32_t)));
    return 0;
}
```

So it appears that the softdevice is present and enabled which I thought I deliberately disabled with the above settings in `CUSTOM_TARGET`?

0xc0170, 2018-02-12:
@ARMmbed/team-nordic Please review

drahnr, 2018-02-12:
Also: If `size` is `4*N+[1..3]` the last `[1..3]` bytes will be lost (given `sizeof(uint32_t) == 4` )

drahnr, 2018-02-13:
The above never happens, since that is prevented by the `FlashIAP` interface to be only complete blocks, aligned to the block layout.

closed 2018-02-13 by drahnr

drahnr, 2018-02-13:
Also: I had a wrong expectation of writing arbitrary addresses and FlashIAP takeing care of the nitty gritty details of alignment.

0xc0170, 2018-02-19:
> Also: I had a wrong expectation of writing arbitrary addresses and FlashIAP takeing care of the nitty gritty details of alignment.

Can you provide details? Should it take care of it?

drahnr, 2018-02-19:
My expectation was that I can write a byte array of arbitrary length to an arbitrary address (given both are valid and fit into the flash). This was the basic assumption which I assumed which was obviously wrong. There is nothing really to be taken care of other than clearly stating this behaviour.

0xc0170, 2018-02-20:
> My expectation was that I can write a byte array of arbitrary length to an arbitrary address (given both are valid and fit into the flash). This was the basic assumption which I assumed which was obviously wrong. 

https://github.com/ARMmbed/mbed-os/blob/master/drivers/FlashIAP.h#L79 - there are requirements captured there. Can you send PR that would state the behavior (what is missing there) ?

drahnr, 2018-02-26:
Well, there is not much to say in my defence, the statements are clear and concise, I just plain old overlooked them.