ARMmbed/mbed-os #6072: FAT Filesystem error code maps to incorrect errors

kegilbert, 2018-02-12:
The fat error remap case structure in the FATFileSystem class maps FR_INT_ERR, FR_MKFS_ABORTED, FR_TIMEOUT, and the default case to the a bad file error code which doesn't make much sense for the either a default or for the above errors.

Initially saw this during CI testing when I would receive the bad file error on format operations.

https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L66

geky, 2018-02-15:
@0xc0170, quick question, do we consider changing error codes to match documentation acceptable for patch releases? My gut says no, but at the same time these may be considered bugs?

0xc0170, 2018-02-16:
> @0xc0170, quick question, do we consider changing error codes to match documentation acceptable for patch releases? My gut says no, but at the same time these may be considered bugs?

 If API specifies error codes and those are correct, it is a bug in the implementation. Should be fine for patch release (assuming it has clear commit msg and gets to the release notes)

kegilbert, 2018-02-16:
Additionally should  https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/FATFileSystem.cpp#L64  be EINVAL rather than ENOEXEC?

geky, 2018-02-16:
Looking at this enum, here's what I'm thinking for the mapping:

| ChanFS                  | was        | should be  | ChanFS description                                               |
| ----------------------- | ---------- | ---------- | ---------------------------------------------------------------- |
| FR_OK                   | 0          | 0          | Succeeded                                                        |
| FR_DISK_ERR             | EIO        | EIO        | A hard error occurred in the low level disk I/O layer            |
| FR_INT_ERR              | ~EBADF~    | -1         | Assertion failed                                                 |
| FR_NOT_READY            | EIO        | EIO        | The physical drive cannot work                                   |
| FR_NO_FILE              | ENOENT     | ENOENT     | Could not find the file                                          |
| FR_NO_PATH              | ~ENOENT~   | ENOTDIR    | Could not find the path                                          |
| FR_INVALID_NAME         | ~ENOENT~   | EINVAL     | The path name format is invalid                                  |
| FR_DENIED               | EACCES     | EACCES     | Access denied due to prohibited access or directory full         |
| FR_EXIST                | EEXIST     | EEXIST     | Access denied due to prohibited access                           |
| FR_INVALID_OBJECT       | ~EFAULT~   | EBADF      | The file/directory object is invalid                             |
| FR_WRITE_PROTECTED      | EACCES     | EACCES     | The physical drive is write protected                            |
| FR_INVALID_DRIVE        | ~ENOENT~   | ENODEV     | The logical drive number is invalid                              |
| FR_NOT_ENABLED          | ~ENXIO~    | ENODEV     | The volume has no work area                                      |
| FR_NO_FILESYSTEM        | ~ENOENT~   | EINVAL     | There is no valid FAT volume                                     |
| FR_MKFS_ABORTED         | ~EBADF~    | EIO        | The f_mkfs() aborted due to any problem                          |
| FR_TIMEOUT              | ~EBADF~    | EAGAIN     | Could not get a grant to access the volume within defined period |
| FR_LOCKED               | ~EACCES~   | EBUSY      | The operation is rejected according to the file sharing policy   |
| FR_NOT_ENOUGH_CORE      | ENOMEM     | ENOMEM     | LFN working buffer could not be allocated                        |
| FR_TOO_MANY_OPEN_FILES  | ENFILE     | ENFILE     | Number of open files > FF_FS_LOCK                                |
| FR_INVALID_PARAMETER    | ~ENOEXEC~  | EINVAL     | Given parameter is invalid                                       |

geky, 2018-02-16:
Respective pr: https://github.com/ARMmbed/mbed-os/pull/6120

closed 2018-02-26 by cmonr