ARMmbed/mbed-os #6073: Make it possible to enable C++11 with gnuarmeclipse

amq, 2018-02-12:
### Description
- Type: Enhancement
- Related issue: https://github.com/ARMmbed/mbed-os/issues/4872, https://github.com/ARMmbed/mbed-cli/issues/588
- Priority: Minor

See also https://os.mbed.com/forum/mbed/topic/28583/

`gnuarmeclipse` exporter doesn't support `--profile`, and the only way to enable C++11 is to fork mbed-os. Things are a bit complicated by the fact that `gnuarmeclipse` exports all 3 profiles, while `--profile` is meant to be used with a single one.

I propose two solutions:

- add a new option `--profiles` which accepts a directory. It would be global for the whole `mbed-cli`
- make `gnuarmeclipse` treat `--profile` as a directory

Why don't I like `eclipse_gcc_arm` which does support `--profile`?

- indexing is much slower and doesn't resolve many things, not sure why
- it creates a clutter `eclipse-extras` directory
- I'm not able to quickly switch between profiles

Why is it that hard to enable C++11 in project settings?

- I tend to export quite often, which is already hurting productivity quite a bit. I'd like to avoid additional delays and re-indexing. For me it is a strong enough reason to still keep a fork of mbed-os, despite being *that* close to being able to use upstream

JojoS62, 2018-02-13:
I've written the mcuxpresso exporter and had a problem with creating other build configurations as release or debug. These names seem to be hard coded in the eclipse MCUXpresso plugin. The base for this exporter was the gnuarmeclipse, there it works to create additional profiles:
```
mbed export -m LPC4088 -i gnuarmeclipse --profile mbed-os/tools/profiles/myrelease-c11.json
```
myrelease-c11.json can be a copy of the standard release profile with your modified settings. You will get the three built-in profiles release, debug and develop plus the user profiler myrelease-c11. 
Writing the exporter took a lot of time by reverse engeneering the necessary conversions for the ecplipse project file, currently I don't have time to fix this (if it is even possible).
Another workaround is to change the mbed-os/tools/profiles/release.json to the required settings.

amq, 2018-02-18:
@JojoS62 this doesn't work for me, I'm only getting the 3 default profiles.

```
cp mbed-os/tools/profiles/release.json \
   myrelease-c11.json

mbed export -m LPC4088 -i gnuarmeclipse --profile myrelease-c11.json
```

This works, but it doesn't solve my problem, since it requires a fork:

```
cp mbed-os/tools/profiles/release.json \
   mbed-os/tools/profiles/myrelease-c11.json

mbed export -m LPC4088 -i gnuarmeclipse
```

I propose a third solution: if user specified `--profile`, export only that single profile. Seems to be simple and intuitive.

JojoS62, 2018-02-18:
you're right, my suggestion works only when the additional profile is found int the tools/profiles directory.  I would consider this as a bug and if this is fixed it may solve your problem.
Creating _only_ the specified profile I would like when I can add multiple profiles, switching between different profiles is a great thing in the IDE.

bmcdonnell-ionx, 2018-04-13:
My current work-around is to change to C++11 manually via the MCUXpresso GUI after I export.

- Right click
- Project Properties
- C/C++ Build
- Settings
    - Select `[ All Configurations ]` in the "Configuration" pull-down
- Tool Settings tab
- MCU C++ Compiler
- Dialect
- Language standard
    - GNU C++11 (`-std=gnu++11`)

0xc0170, 2018-05-22:
@ARMmbed/mbed-os-tools Shall we close this as won't fix as this is unsupported ?

0xc0170, 2019-08-14:
We enabled C++14, this can be closed now

closed 2019-08-14 by 0xc0170