ARMmbed/mbed-os #6077: FlashIAP does not deal with erasing sectors correctly

LiyouZhou, 2018-02-13:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
Any STM based targets with Variable internal flash sector size

**Toolchain:**
Any

**mbed-os sha:**
caeaa49d6 Merge pull request #5954 from ARMmbed/release-candidate

**Expected behavior**
FlashIAP::erase allows the user to erase multiple sectors with one call.

On STM F439 (ODIN) sector 12 starting from 0x080e0000 have size 128k, sector 13 starting from 0x08100000 have size 16k. I want to erase these two sectors so I call

FlashIAP::erase(0x080e0000, (128+16)*1024)

**Actual behavior**
The impelmentation cannot deal with variable sector sizes and returns -1 on alignment check.

**Steps to reproduce**
compile the following code:
`mbed compile -m UBLOX_EVK_ODIN_W2 -t GCC_ARM` and run.
```
#include "mbed.h"

int main()
{
    FlashIAP flash;

    flash.init();

    size_t sec_1_start_addr = flash.get_flash_start() + 0xe0000;
    size_t sec_1_size = flash.get_sector_size(sec_1_start_addr);
    size_t sec_2_start_addr = sec_1_start_addr + sec_1_size;
    size_t sec_2_size = flash.get_sector_size(sec_2_start_addr);

    int ret = flash.erase(sec_1_start_addr, sec_1_size + sec_2_size);
    printf("ret %i\r\n", ret);

    ret = flash.erase(sec_1_start_addr, sec_1_size);
    printf("ret %i\r\n", ret);

    ret = flash.erase(sec_2_start_addr, sec_2_size);
    printf("ret %i\r\n", ret);

    return 0;
}
```

The first call fails while the subsequent calls succeed.

## RCA

In FlashIAP.cpp https://github.com/ARMmbed/mbed-os/blob/master/drivers/FlashIAP.cpp#L118-L144
The algorithm tries to erase sector by sector but the alignment check is still done on the overall erase size. This will not work on variable sector sizes.

The API does not provide a way for user to get the correct erase size. User is forced to doing erasing secotor by sector at the top level thus defeating the purpose of the FlashIAP::erase API.

0xc0170, 2018-02-13:
@SenRamakri Please review

0xc0170, 2018-02-13:
@LiyouZhou Do you have any proposal how to fix this?

As the address + size to erase might overlap to different sized sectors, the check you referenced fails (not aligned). 
We can add a check for sector sizes over the entire size to be erased to check how many different sizes are involved and do checks per each one found.

LiyouZhou, 2018-02-13:
This is how i fixed it https://github.com/ARMmbed/mbed-bootloader-internal/blob/master/source/active_application.cpp#L190-L237

If you do pretty much the same thing in flashiap it would work.

davidsaada, 2018-06-18:
@0xc0170 - This issue can be closed now. [PR #6140](https://github.com/ARMmbed/mbed-os/pull/6140) fixed it.

closed 2018-06-19 by 0xc0170