ARMmbed/mbed-os #6079: Stm32l4 shutdown bug

ivanccu, 2018-02-13:
### Description
- Type: Bug  | Question
- Related issue: `Sleep and deepsleep modes work differently on two equal boards`
- Priority: Blocker 

There is no shutdown functionality in mbed api, so to achieve this I use HAL code mixed whit mbed api (see the link)
https://os.mbed.com/questions/79533/Sleep-and-deepsleep-modes-work-different/

and unexpected behavior appears

---------------------------------------------------------------
## Bug

**Target**
nucleo smt32L4

**Toolchain:**
mbed online

**Expected behavior**
After reset the board or reprograming the registers must have the default values
**Actual behavior**
After trying shutdown mode the board consumption increases, in spite of reset the board or reprograming.
**Steps to reproduce**
https://os.mbed.com/questions/79533/Sleep-and-deepsleep-modes-work-different/

-----------------------------------------------------------------

## Question

**How to?**

Return to the values ​​of original registers, since deleting the chip before reprogramming does not work

**How to?** 
Enter in shutdown mode (stm32L4)  with mbed

MarceloSalazar, 2018-02-13:
Requirement for shutdown API logged here: IOTPREQ-2538

However, from what I can understand, there is a bug in the HAL/initialization code that is not initializing the system correctly, hence requires investigation.

0xc0170, 2018-02-13:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-02-15:
> I use HAL code mixed with mbed api and unexpected behavior appears

We don't encourage people to do this mix. I recommend you to use only mbed API...

LMESTM, 2018-02-16:
@ivanccu  not easy to help remotely on such complex use case with both HW and SW implications so would be good to take the points 1 by 1.  Let me try to propose for now a few directions to look at
- Can you check the stlink version of your 3 boards and see if they are the same.  Upgrade them all to the latest version (http://www.st.com/en/development-tools/stsw-link007.html)
- Verify that all resistors, jumpers are equally positioned on all boards
- when testing deepsleep and shutdown modes, and measuring power consumption, I would advise to not use the Serial link to PC (Maybe use LEDs instead or simply enter / exit shutdown mode in your program (no sleep / deepsleep in a first step)
 -maybe have a look at online support documentation for low power modes (like http://www.st.com/content/ccc/resource/training/technical/product_training/ce/57/a3/86/7a/3d/4d/87/STM32L4_System_Power.pdf/files/STM32L4_System_Power.pdf/jcr:content/translations/en.STM32L4_System_Power.pdf, see Debug control register explanation for instance). 

From your description, I don't clearly understand what happens after shutdown mode. How much is the power consumption ?

Hope this helps
Laurent

ivanccu, 2018-02-16:
@bcostm the reason why I was trying to mix code is because that function is not collected in the mbed API.
If there was such a function in the API or if they updated the API taking into account ultralow modes, it would be perfect, but as it did not exist I was still forced to touch that dark part of the code
@LMESTM 
I followed the steps you told me and I tell you what I have obtained, even so I will explain a little the operation of the program because I think that I have generated doubts

> Can you check the stlink version of your 3 boards and see if they are the same. Upgrade them all to the latest version

the three board had the same version in the stlink, anyway I have forced an update to the same version just in case.

> Verify that all resistors, jumpers are equally positioned on all boards

all the jumpers have been checked and had the same position in the 3 boards

> when testing deepsleep and shutdown modes, and measuring power consumption, I would advise to not use the Serial link to PC (Maybe use LEDs instead or simply enter / exit shutdown mode in your program (no sleep / deepsleep in a first step)

the code used both things (serial communication and led warning). I have tried again removing the serial communication part and the consumption problem persists.
Even so the serial port in theory should not affect since it is only used when the equipment is awake.
In the code, the first thing I do is to be awake by increasing a variable (timestamp) and I only go to sleep when I press a button. If I put the button back I wake up and continue the count. Testing in this way the function sleep, deepsleep and shutdown (in shutdown the account must be restarted) and the consumption of each state

> From your description, I don't clearly understand what happens after shutdown mode. How much is the power consumption ?

The problem is that despite completely erasing the chip with the stlink and reprogramming the board with a dummy program (a led that blinks every second and goes to sleep in the highest mode (sleep () function)
My consumption goes from being 35uA (as in the data sheet and checked with that same dummy program) to being 80-90 uA.

I trust that now the problem is clearer, it's like something is stuck consuming,
Thanks for your help.
Greetings,
Ivan

LMESTM, 2018-02-16:
@ivanccu ok thanks the explanation about power consumption before and after shutdown is more clear now. Not easy to figure out for now, I'd need some lower power expert guidance internally.  One thing you way want to try is to compare registers content with your debugger on the backup power domain, which are the only registers that will not be reset. Most probably, you have activated a different wake-up source or mode inside your  shutdown code (so most probably in backup domain) that needs to be reset at the next boot.  

From reading your code, the             HAL_PWR_EnableWakeUpPin(PWR_WAKEUP_PIN2_LOW);
is called for shutdown and also in in the pressed() callback, but it 's never disabled even after shutdown exit. isn't that a visible difference of your SW before and after shutdown ? 

note: I mentioned removing serial link to avoid power consumption leakage through the serial pins.  I asked for stlink upgrade for similar reason to avoid leakage thru JTAG debug pins

jeromecoutant, 2018-04-25:
Hi Ivan
Is there any updates for this issue ?
Thx

ivanccu, 2018-04-26:
Greetings the truth is that I have already rejected that board because I may have something wrong and now I use another board (same model).
With the new realease of mbed I get to sleep and to deepsleep (now I think it's called sleep auto) the same consumption 1,2 uA. so it seems that even the 400 nA are not achieved.
From what I have seen inside the code of mbed, they use the function HAL_PWREx_EnterSTOP2Mode but  the consumption does not lower to 400nA indicates that something is not disabled correctly

0xc0170, 2018-06-18:
@jeromecoutant Latest changes to low power ticker, tickless and sleep manager - should all these address the issue here (shutdown mode is a different issue) ?

jeromecoutant, 2019-10-11:
@adbridge 
As there is no action on this question,
I think we can close it ?
Thx

closed 2019-10-11 by adbridge