ARMmbed/mbed-os #6081: Add shutdown mode to sleep api

ivanccu, 2018-02-13:
### Description
- Type: Enhancement 
- Related issue: `Shutdown api`
- Priority: Major 

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
Currently it is starting to work in the industrial sector with the internet of things. 
One of the characteristics of IoT is the low consumption. To achieve this low consumption, manufacturers already have microcontrollers such as the STM32L4 capable of entering ultra-low consumption mode (shutdown), sacrificing information not stored in memory.
This utility is very useful at the moment but, in spite of the fact that in this case ST has implemented this functionality for its microcontroller, this function does not exist in the mbed library.
With the current library (sleep) you can only reach two operating modes (sleep and deepsleep) that have higher consumption than shutdown.
Therefore if you need to use the shutdown function in your project you need to go down to the HAL level and try to do it on your own. Possibly resulting errors since going down to the HAL layer implies losing the thread safety

**Suggested enhancement**
Add shutdown mode to de mbed library
**Pros**
-more easy to use for the user
-mbed will be more itnerested for the market, specially on Iot devices

**Cons**

 -invalid function for some microcontrollers
-----------------------------------------------------------------

LMESTM, 2018-02-13:
Please consider adding RTC::alarm service together with shutdown

closed 2018-12-19 by ciarmcom

jeromecoutant, 2018-12-20:
Hi
Why there is no explanation for issue closing ?
Thx

0xc0170, 2018-12-20:
This is being tracked internally via another ticket. Thus this was closed and original kept (it's not on Github as an issue).

@donatieng can correct me.

cmonr, 2018-12-20:
Either way, if an issue was closed because it was duplicated, a message indicating what it's a duplicate _of_ should be provided.