ARMmbed/mbed-os #6089: mbed export -i IAR -m NUCLEO_L443RC not working with mbed-os-5

kmkuo, 2018-02-14:
Target
NUCLEO_L443RC

Toolchain:
IAR

Toolchain version: 6.4.0

mbed-cli version:
1.4.0

Steps to reproduce
mbed export -i IAR -m NUCLEO_L443RC

0xc0170, 2018-02-14:
> mbed export -i IAR -m NUCLEO_L443RC not working with mbed-os-5 #6089

What does not working mean? Does it fail to export, fail to build?

running locally, `NUCLEO_L443RC` is not supported by IAR exporter.

@ARMmbed/team-st-mcd Please review

bcostm, 2018-02-14:
The NUCLEO_L433RC platform is not supported (it is not present in targets.json file).

But there is instead the NUCLEO_L433RC_P platform where IAR is supported (normally).

bcostm, 2018-02-14:
The NUCLEO_L443RC platform does not exist...

kmkuo, 2018-02-15:
It's show not supported by IAR.

usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: NUCLEO_L443RC not supported by iar
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.

bcostm, 2018-02-15:
Which mbed version do you use ?

On my side:
```
$ mbed --version
1.4.0

$ mbed export -i IAR -m NUCLEO_L443RC
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S [{matrix,ides}]] [-E] [--source SOURCE_DIR]
                  [-D MACROS] [--profile PROFILE] [--update-packs]
                  [--app-config APP_CONFIG]
project.py: error: argument -m/--mcu: NUCLEO_L443RC is not a supported MCU. Supported MCUs are:
...
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
```

same with:
```
$ mbed export -i IAR -m NUCLEO_XXXXXXXXXX
...
project.py: error: argument -m/--mcu: NUCLEO_XXXXXXXXXX is not a supported MCU. Supported MCUs are:
...
```

kmkuo, 2018-02-16:
my mbed-cli version is 1.4.0

kmkuo, 2018-02-21:
Dose any one know how to add IAR  to support NUCLEO_L443RC manually?

Thank you.

0xc0170, 2018-06-18:
@kmkuo As asked previously, this target is not supported and not in the codebase? 

This fixes the issue https://github.com/ARMmbed/mbed-os/pull/7020 . The mcu definition was added for IAR.

closed 2018-06-18 by 0xc0170