ARMmbed/mbed-os #6090: LittleFS - root directory mkdir

moranpeker, 2018-02-14:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Major 

---------------------------------------------------------------
## Bug

**please follow this mkdir scenario:**

lfs info: mkdir("pri/work1", 0x180)
lfs info: mkdir -> 0

[DBG ][PAL ]: getRootPath = /sd   (** try to use mkdir to create a getRootPath  dir. this is my printf to the root directory path)

lfs info: mkdir("", 0x180)    (why it its empty and not "/sd" ? (root directory path))
lfs info: mkdir -> 0

**Expected behavior**
mkdir should fall with error of FS name already exist

**Actual behavior**
succeed to create "" dir instead of /sd

0xc0170, 2018-02-14:
@geky Please review

geky, 2018-02-15:
Ah, in your application is this due to these two calls?
``` cpp
mkdir("/sd/pri/work1", 0x180);
mkdir("/sd", 0x180);
```

If that's the case the problem is a bit higher than the filesystem and is in the retarget layer. I'll see if I can track it down.

moranpeker, 2018-02-18:
yes this is exactly what we are doing .

geky, 2018-02-21:
Huh, actually https://github.com/ARMmbed/mbed-os/pull/6156 fixes this as a side-effect.

With https://github.com/ARMmbed/mbed-os/pull/6156, this will now assert. Are you relying on the ENOENT error code? I'll see if I can get the error code in.

geky, 2018-02-22:
No, nevermind, this needs to be fixed elsewhere, still looking...

geky, 2018-02-22:
Ok, looks like the issue was in the littlefs layer, lfs_mkdir("/") wasn't erroring correctly.

I've pushed a patch here: https://github.com/ARMmbed/mbed-os/pull/6177. Does https://github.com/ARMmbed/mbed-os/pull/6177 fix your issue?

closed 2018-02-26 by cmonr