ARMmbed/mbed-os #6091: Critical Sections when "FEATURE_UVISOR" not defined

paul-adam, 2018-02-14:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

In a source code without "FEATURE_UVISOR" defined:
In file "mbed_critical.c" there are the lines (from line 68):
// FIXME
#ifndef   FEATURE_UVISOR
    if (interrupt_enable_counter > 0) {
        MBED_ASSERT(interrupts_disabled);
    }
#else
thus the function "core_util_critical_section_enter" cannot be called twice without  calling "core_util_critical_section_exit".

Or the function "Timer::start()" calls inside a critical section the function "ticker_read_us":
void Timer::start() {
    core_util_critical_section_enter();
    if (!_running) {
        if(_lock_deepsleep) {
            sleep_manager_lock_deep_sleep();
        }
        _start = ticker_read_us(_ticker_data);
        _running = 1;
    }
    core_util_critical_section_exit();
}

Function "ticker_read_us" itself calls "core_util_critical_section_enter":
us_timestamp_t ticker_read_us(const ticker_data_t *const ticker)
{
    initialize(ticker);

    core_util_critical_section_enter();
    update_present_time(ticker);
    core_util_critical_section_exit();

    return ticker->queue->present_time;
}

This does not work...

**Target**
B_L475E_IOT01A

**Toolchain:**
GCC_ARM

**Toolchain version:**
?

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
?

**DAPLink version:**
?

**Expected behavior**
In file "mbed_critical.c" the code should be changed to  (from line 68):
// FIXME
#ifndef   FEATURE_UVISOR
    if (interrupt_enable_counter > 1) {
        MBED_ASSERT(interrupts_disabled);
    }
#else

**Actual behavior**
Target executes "mbed_die". 

**Steps to reproduce**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2018-02-14:
@scartmell-arm @avolinski  Please review

scartmell-arm, 2018-02-22:
I'm not sure what the issue is. The highlighted code:

```c++
if (interrupt_enable_counter > 0) {
	MBED_ASSERT(interrupts_disabled);
}
```

is a debug only assert which checks that interrupts are already disabled when entering the function on a nested call, which they should be. The counter is incremented at the end of the function so at this point `counter > 0` is correct.

scartmell-arm, 2018-02-22:
So, the issue you're having is that while debugging `core_util_critical_section_enter()` is failing to disable interrupts correctly, causing an assertion?

Does this happen on any other boards if you have any available?

0xc0170, 2018-06-18:
@paul-adam  Any update for the issue?

SenRamakri, 2018-12-04:
Hi @paul-adam - We think this is no longer an issue, please see the latest implementation in mbed_Critical.c. Please let us know if you still have questions.

closed 2018-12-05 by 0xc0170