ARMmbed/mbed-os #6100: equeue_platform.h cannot compile without RTOS due to non-necessary included headers

janjongboom, 2018-02-15:
### Description
- Enhancement

## Enhancement

**Reason to enhance or problem with existing solution**

EventQueue can be used without RTOS fine, just run it in the main thread. However, equeue_platform.h references these two header files:

```
#include "cmsis_os2.h"
#include "mbed_rtos_storage.h"
```

Actually this file compiles fine if you don't include those files and then eventqueue still works properly.

**Suggested enhancement**

Wrap those includes in an ifdef `RTOS_PRESENT` (or what the macro for that is).

**Pros**

Can run EventQueue in a non-RTOS environment, which would be necessary to run our LoRaWAN stack on the DISCO LORAWAN board.

**Cons**

-

0xc0170, 2018-02-15:
@geky Please review

geky, 2018-02-15:
Good catch, thanks for reporting, looks like a simple regression that slipped in.

This should fix it up real good like https://github.com/ARMmbed/mbed-os/pull/6110

closed 2018-02-16 by cmonr