ARMmbed/mbed-os #6101: Are there any plans to support STM32 H7 boards?

peeyushpashine, 2018-02-15:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type:  Question
- Priority: Minor

---------------------------------------------------------------
-----------------------------------------------------------------

## Question

Are there any plans to support STM32 H7 boards?

bcostm, 2018-02-15:
Hi,
We have started the NUCLEO_H743ZI SDK but it is not ready yet. No official date for the moment.

0xc0170, 2018-02-15:
@peeyushpashine If that answers your question, please close this issue

closed 2018-02-15 by peeyushpashine

LMESTM, 2018-02-16:
@peeyushpashine  I would add that community contributions are more than welcome, so if you want to get this support right now and propose the H7 add-on support, we'd be glad to review your PRs.

jeromecoutant, 2018-02-16:
Hi
I can share my local working view:
https://github.com/jeromecoutant/mbed/tree/DEV_NUCLEO_H743ZI

I will not work on it for some time, so community contributions are welcome !

OmerRosler, 2018-10-09:
Is there any change regarding this board?
We really really want to use and no mbed support is a deal breaker

seshsubb, 2018-10-09:
Hello @jeromecoutant this link is unreachable now - https://github.com/jeromecoutant/mbed/tree/DEV_NUCLEO_H743ZI. Can you please check this? 

We plan to use STM32H750. Can you confirm if the mbed SDK for DEV_NUCLEO_H743ZI will be usable for STM32H750 with few minimal changes, or will there be a lot of changes?

LMESTM, 2018-10-11:
@OmerRosler @seshsubb and any other interested 
The new dates are being discussed internally. Can you share by email more details about your projects (my email appears in my commits sign-off) so that we can discuss timelines and options.

jeromecoutant, 2018-10-11:
Hi

> Can you confirm if the mbed SDK for DEV_NUCLEO_H743ZI will be usable for STM32H750 with few minimal changes,

Yes, it should be easy

seshsubb, 2018-10-12:
> Hi
> 
> > Can you confirm if the mbed SDK for DEV_NUCLEO_H743ZI will be usable for STM32H750 with few minimal changes,
> 
> Yes, it should be easy

Thanks @jeromecoutant @LMESTM. The link is not accessible - https://github.com/jeromecoutant/mbed/tree/DEV_NUCLEO_H743ZI. We had taken a local copy sometime back. Hopefully it is good to use  for the time being.

shlomow, 2018-12-14:
@jeromecoutant 
Are there any news? We are before the end of the year and my team is also interested in mbed for stm32h7. Thanks!

jeromecoutant, 2018-12-14:
@shlomow you should check #9056 
Regards,