ARMmbed/mbed-os #6103: Missing python3 dependencies after upgrading to Mbed OS 5.7.5

teetak01, 2018-02-15:
### Description
- Type: Bug
- Priority: Major
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**mbed-os sha:**
Mbed OS 5.7.5

```
stdout: [mbed] Auto-installing missing Python modules...
 
stderr: [mbed] WARNING: Could not find mbed program in current path "XX".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
Traceback (most recent call last):
  File "/XX/mbed-os/tools/make.py", line 43, in <module>
    from tools.options import get_default_options_parser
  File "/XX/mbed-os/tools/options.py", line 21, in <module>
    from tools.toolchains import TOOLCHAINS
  File "/XX/mbed-os/tools/toolchains/__init__.py", line 36, in <module>
    from tools.memap import MemapParser
  File "/XX/mbed-os/tools/memap.py", line 16, in <module>
    from tools.arm_pack_manager import Cache
  File "/XX/mbed-os/tools/arm_pack_manager/__init__.py", line 18, in <module>
    from tools.flash_algo import PackFlashAlgo
  File "/XX/mbed-os/tools/flash_algo/__init__.py", line 30, in <module>
    from elftools.common.py3compat import bytes2str

ImportError: No module named elftools.common.py3compat
```

This started to fail in one build-environment which uses Python3 virtual environment. I have not seen this in other environments, but the breaking change was the update from Mbed OS 5.7.4=>5.7.5

Which component do I need to install for this to work?

0xc0170, 2018-02-15:
It should be `pyelftools` that is in requirements. @teetak01 Was it installed ?

@theotherjimmy Please review

teetak01, 2018-02-15:
Thanks @0xc0170 , installing pyelftools fixed it for me. I thought to raise an issue on this as others might have noticed something similar too.

closed 2018-02-15 by teetak01