ARMmbed/mbed-os #6105: STM32F4 change stdio pins

pilotak, 2018-02-15:
When i try to change default STDIO pins through mbed.json as per STM32 [example](https://developer-sjc-indigo-border.mbed.org/teams/ST/wiki/STDIO)
```json
{
  "config": {
    "stdio_uart_tx": {
      "value": "PA_2"
    },
    "stdio_uart_rx": {
      "value": "PA_3"
    }
  }
}
```
It doesn't work for me, which I think it would not work because macro ` MBED_CONF_TARGET_STDIO_UART_TX` in `PinNames.h` is saying it should be in **target_overrides** section of `mbed.json` file. So the macro should be called be called `MBED_CONF_APP_STDIO_UART_TX` in `PinNames.h` if i'm right.
**or**
to leave it as it is I need to add  to config section `targets/target.json` **stdio_uart_tx** and **stdio_uart_rx**
```json
"NUCLEO_F412ZG": {
        "inherits": ["FAMILY_STM32"],
        "supported_form_factors": ["ARDUINO", "MORPHO"],
        "core": "Cortex-M4F",
        "extra_labels_add": ["STM32F4", "STM32F412xG", "STM32F412ZG"],
        "config": {
            "stdio_uart_tx": {
                "help": "Value PD_8",
                "value": "PD_8"
            },
            "stdio_uart_rx": {
                "help": "Value PD_9",
                "value": "PD_9"
            },
            "clock_source": {
                "help": "Mask value : USE_PLL_HSE_EXTC | USE_PLL_HSE_XTAL (need HW patch) | USE_PLL_HSI",
                "value": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
                "macro_name": "CLOCK_SOURCE"
            }
        },
        "detect_code": ["0826"],
        "macros_add": ["USB_STM_HAL", "USBHOST_OTHER"],
        "device_has_add": ["CAN", "LOWPOWERTIMER", "SERIAL_ASYNCH", "SERIAL_FC", "TRNG", "FLASH"],
        "release_versions": ["2", "5"],
        "device_name": "STM32F412ZG",
        "bootloader_supported": true
    },
```
and change `mbed.json` to 
```json
{
    "target_overrides": {
        "NUCLEO_F412ZG": {
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```

It than works as expected

jeromecoutant, 2018-02-15:
Hi

Please check #6041 

Regards,

pilotak, 2018-02-15:
Ok did't see that one. Good someone else spotted it too and came to same result as me

closed 2018-02-15 by pilotak