ARMmbed/mbed-os #6107: Shared queue hangs when called from interrupt

marcemmers, 2018-02-15:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**mbed-os sha:** b8d218038bcd8c2ad447d91ab94c927118c1ad61
Still present in 5.7.5

Using the function mbed_event_queue() to queue a function call in interrupt context fails when used directly like this:
```
isr_context()
{
    mbed_event_queue()->call(func);
}
```

The problem comes from the thread.get_state() call where a mutex is getting locked which is not allowed in isr context. When the queue ptr is first stored in a variable and then used to call the "call" function it works without problems.

Maybe it would be better to use a separate flag to check if the thread has been started because you don't need to get the thread state every time:

```
<osPriority Priority, size_t QueueSize, size_t StackSize>
EventQueue *do_shared_event_queue_with_thread()
{
    static uint64_t queue_buffer[QueueSize / sizeof(uint64_t)];
    static EventQueue queue(sizeof queue_buffer, (unsigned char *) queue_buffer);

    static uint64_t stack[StackSize / sizeof(uint64_t)];
    static Thread thread(Priority, StackSize, (unsigned char *) stack);

    static bool initialized = false;

    if (!initialized) {
        osStatus status = thread.start(callback(&queue, &EventQueue::dispatch_forever));
        MBED_ASSERT(status == osOK);
        if (status != osOK) {
            return NULL;
        }
        initialized = true;
    }

    return &queue;
}
```

geky, 2018-02-15:
Oh that is interesting, I think your fix sounds correct. @kjbracey-arm thoughts?

We may still need to synchronize access to initialized, not sure. But we can use a critical section in this case.

kjbracey-arm, 2018-02-16:
BTW, I'm not sure whether the proposed fix in the descriptions works because of the remaining `static Thread` and `EventQueue` objects. I know the construction for static non-POD objects like that is thread-safe, but are the locks used to achieve that also interrupt-safe?

geky, 2018-02-16:
Interesting. You could maintain irq safety by keeping pointers to the local static objects. Though I see your point that it may be better to just not allow `mbed_event_queue()` in irq.

Since IRQ safety is the core of the event queues I went ahead and added a note: https://github.com/ARMmbed/mbed-os/pull/6118

@marcemmers, what are your thoughts on this?

marcemmers, 2018-02-19:
Its not really that big a problem if it needs to be stored first in thread mode but it was indeed never mentioned. Good to know that is solved already!

However, if you use the shared queue from main thread it is possible to call `mbed_event_queue()` directly.

It might be an idea for later on to start the shared thread during startup. The biggest chunk of memory is already used for the queue_buffer and stack and will be allocated even if the queue is never used so the additional memory of the actual thread and queue objects wont be as significant.

kjbracey-arm, 2018-02-19:
If `mbed_event_queue()` is never referenced, the memory will not be allocated - the function and its static data won't be linked in.  Or that's the intent and reasoning behind the design, anyway. (Hope you're not seeing anything contrary to that).

marcemmers, 2018-02-20:
@kjbracey-arm you are right ofcourse, didn't really think that through.

I think this can be closed then?

closed 2018-02-20 by kjbracey-arm