ARMmbed/mbed-os #6111: dir_read function return unexist file/directory

moranpeker, 2018-02-15:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#abc`
- Priority:  Major 

---------------------------------------------------------------
## Bug

**please follow this scenario:** _(i added some prints in addition to the lfs prints)_

lfs info: dir_read(0x2001a2e8, 0x2000add4)
lfs info: dir_read -> 1
_pal_print - readdir function output:   test_f1 (this is the name of the file inside the dir)_
_pal_print - full path:   /sd/pri/dir1/test_f1_ 
_pal_print -  file type:   5 (== DT_REG,     ///< This is a regular file.)_
lfs info: remove("pri/dir1/test_f1")
lfs info: remove -> 0
lfs info: dir_read(0x2001a2e8, 0x2000add4)
lfs info: dir_read -> 1
__pal_print - readdir function output:   test_f19aJ_ 
_pal_print - full path:   /sd/pri/dir1/test_f19aJ_ 
_pal_print -  file type:    3  (== DT_DIR,     ///< This is a directory.)__
lfs info: dir_open(0x0, "pri/dir1/test_f19aJ")
lfs info: dir_open -> -2  (== FS ERROR NO PATH)
lfs info: remove("pri/dir1/test_f19aJ")
lfs info: remove -> -2

**Actual behavior**
First, dir_read return an exist file first . file type is regular file. we remove it successfully.
at the second time dir_read  return unexist file - (name test_f19aJ ). it's type is directory. try to open it and dir_open return -2 because its not exist.

**Expected behavior**
dir_read function must return only existing files.

geky, 2018-02-15:
Do you have a link to the source code that is causing this problem? It would help enough to know exactly how functions are getting called.

The trace helps, but from this perspective all I can tell is that the null pointer (0x0) passed to dir_open looks very suspicious.

geky, 2018-02-15:
Actually, looks like the NULL pointer print in dir_open is a mistake on my side.

I'm having a difficult time reproducing the failure, what is the layout of the directory you start with?

moranpeker, 2018-02-18:
yes . the bug is coming from : 
https://github.com/ARMmbed/mbed-client-pal/blob/master/Source/Port/Reference-Impl/OS_Specific/mbedOS/Storage/FileSystem/pal_plat_fileSystem.cpp#L267

i added prints at the trace for full path 
for example :  "pri/dir1/test_f19aJ"
it starts from root path "/sd"

geky, 2018-02-22:
I'm going to need the full directory layout before the error happened. I can create the test_f1 file and test_f19aJ directory and remove them without errors. Are you able to send me a copy/image of the SD card?

Or, how do I run the test that is failing? Is this coming from the PAL `directoryTests`?
https://github.com/ARMmbed/mbed-client-pal/blob/68107ad078d92c7a880142442465d3bd33f69e2d/Test/PAL_Modules/FileSystem/pal_fileSystem_test.c#L431

yossi2le, 2018-07-01:
@moranpeker 
Hi I have tried to reproduce the issue but to no avail.
Please consider closing this issue.
Thanks

closed 2018-07-01 by moranpeker