ARMmbed/mbed-os #6122: Instantiating OdinWiFiInterface seems to be changing Thread behavior

dmelo, 2018-02-16:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-g++ (Fedora 7.1.0-2.fc26) 7.1.0

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
caeaa49d6 (HEAD, tag: mbed-os-5.7.4, tag: latest) Merge pull request #5954 from ARMmbed/release-candidate

**DAPLink version:**

**Expected behavior**
I have two threads, the first runs i2cslave and just prints what it gets from the master and the second thread prints a simple string and then sleeps for 1 second. Both threads are in infinite loop.

Execution 1 - When I run this program, everything runs just fine.

Execution 2 - When I add an global OdinWiFiInterface instance, I expected both thread to keep running and showing the printf results on the serial.

**Actual behavior**

On execution 2, the i2C slave thread seems to be starving the second thread. I don't see the printf from the second thread anymore.

**Steps to reproduce**

Run the following code:

```
#include "mbed.h"
#include "rtos.h"
#include "OdinWiFiInterface.h"

#define I2C_ADDRESS 0x25

Thread t1, thread_i2c_slave;

// To reproduce execution 1, comment the line above.
OdinWiFiInterface wifi; // During execution 2, I have instanced a OdinWiFiInterface object.

void func_t1()
{
    for (int i = 0;; i++) {
        printf("FUNC_T1 ============\n%d\n==============\n\n", i);
        Thread::wait(1000);
    }
}

void i2c_slave()
{
    I2CSlave slave(PF_0, PF_1);
    int count = 0;
    char buf[10];
    char msg[] = "Slave!";

    slave.address(I2C_ADDRESS);
    while (1) {
        int i = slave.receive();
        count++;
        switch (i) {
            case I2CSlave::ReadAddressed:
                slave.write(msg, strlen(msg) + 1); // Includes null char
                break;
            case I2CSlave::WriteGeneral:
                slave.read(buf, 10);
                printf("Read G: %s\n", buf);
                break;
            case I2CSlave::WriteAddressed:
                slave.read(buf, 10);
                printf("Read A: strlen %d. ", strlen(buf));
                for (unsigned int c =0; c < strlen(buf); c++) {
                    printf("%X ", buf[c]);
                }
                printf("\n");
                printf("count: %d\n", count);
                break;
        }
        for(int i = 0; i < 10; i++) buf[i] = 0;    // Clear buffer
    }
}

int main()
{
    printf("BEGIN\n");

    printf("starting threads\n");
    t1.start(func_t1);
    thread_i2c_slave.start(i2c_slave);

    t1.join();
    thread_i2c_slave.join();

    return 0;
}

```

Output on the execution 1:
```
BEGIN
starting threads
FUNC_T1 ============
0
==============

Read A: strlen 4. 1 61 62 63 
count: 749070
Read A: strlen 4. 1 61 62 63 
count: 1602034
FUNC_T1 ============
1
==============

Read A: strlen 4. 1 61 62 63 
count: 2386922
Read A: strlen 4. 1 61 62 63 
count: 3237849
FUNC_T1 ============
2
==============

Read A: strlen 4. 1 61 62 63 
count: 4022860
Read A: strlen 4. 1 61 62 63 
count: 4873974
FUNC_T1 ============
3
==============

Read A: strlen 4. 1 61 62 63 
count: 5658975
Read A: strlen 4. 1 61 62 63 
count: 6510014
FUNC_T1 ============
4
==============

Read A: strlen 4. 1 61 62 63 
count: 7294943
Read A: strlen 4. 1 61 62 63 
count: 8145975
FUNC_T1 ============
5
==============

Read A: strlen 4. 1 61 62 63 
count: 8930893
Read A: strlen 4. 1 61 62 63 
count: 9781846
```

Output on execution 2:
```
BEGIN
starting threads
FUNC_T1 ============
0
==============

Read A: strlen 4. 1 61 62 63 
count: 139268
Read A: strlen 4. 1 61 62 63 
count: 952623
Read A: strlen 4. 1 61 62 63 
count: 1778385
Read A: strlen 4. 1 61 62 63 
count: 2602818
Read A: strlen 4. 1 61 62 63 
count: 3429469
Read A: strlen 4. 1 61 62 63 
count: 4258486
Read A: strlen 4. 1 61 62 63 
count: 5087584
Read A: strlen 4. 1 61 62 63 
count: 5912158
Read A: strlen 4. 1 61 62 63 
count: 6738882
Read A: strlen 4. 1 61 62 63 
count: 7568029
Read A: strlen 4. 1 61 62 63 
count: 8397041
Read A: strlen 4. 1 61 62 63 
count: 9225966
Read A: strlen 4. 1 61 62 63 
count: 10054951
Read A: strlen 4. 1 61 62 63 
count: 10882079
Read A: strlen 4. 1 61 62 63 
count: 11709333
Read A: strlen 4. 1 61 62 63 
count: 12536402
Read A: strlen 4. 1 61 62 63 
count: 13363531
Read A: strlen 4. 1 61 62 63 
count: 14190524
Read A: strlen 4. 1 61 62 63 
count: 15017642
Read A: strlen 4. 1 61 62 63 
count: 15844821
Read A: strlen 4. 1 61 62 63 
count: 16671864
```

----------------------------------------------------------------

## Question

How can I make the threads work while using OdinWiFiInterface?

dmelo, 2018-02-19:
Just tested with `mbed-os-5.7.5` and it doesn't run at all when I try to instantiate OdinWiFiInterface. No output on the serial.

With `5.7.5` only run if OdinWiFiInterface is not present (execution 1)

0xc0170, 2018-02-19:
@ARMmbed/team-ublox Please review

andreaslarssonublox, 2018-02-22:
We'll have a look at it. Internal u-blox ref: TE_OM2-159

dmelo, 2018-03-07:
Hi @andreaslarssonublox . Did you get the time to look into this issue? Do you have any news?

andreaslarssonublox, 2018-03-07:
Hi, @dmelo! We have analyzed the issue and are about to test a fix for it. Hopefully we will know if the fix works this or next week.

dmelo, 2018-03-19:
Hi @andreaslarssonublox, does the fix works? If it does, is it possible for me get my hands on it? It would be really helpful.

andreaslarssonublox, 2018-03-29:
Hi, yes we have a fix that will be included in the next driver release.

0xc0170, 2018-05-22:
@andreaslarssonublox Can this be closed (the latest update we have is now on emac feature PR) or was there any earlier? does it fix this issue?

asifrizwanubx, 2018-05-24:
We will target 5.9.1 for this fix. That wasn't included in feature-emac driver release.

0xc0170, 2018-06-19:
> We will target 5.9.1 for this fix. That wasn't included in feature-emac driver release.

@asifrizwanubx Has this been fixed?

dmelo, 2018-06-26:
Does u-blox have any update on this issue?

asifrizwanubx, 2018-06-26:
there will be a drivers release around mid of july which include the this fix as well.

dmelo, 2018-07-20:
Hi @asifrizwanubx . Do you know if the driver will be released soon?

ardysingh, 2018-07-21:
Hi @asifrizwanubx - any eta on the fix? Will it be compatible with mbedOS 5.8?

asifrizwanubx, 2018-07-23:
@dmelo Testing of latest odin drivers taking more time as compare to our expectation due to Access Point and Bridge Feature. I will let you know the PR asap.

asifrizwanubx, 2018-07-23:
@dmelo done.

dmelo, 2018-07-23:
Thank you @asifrizwanubx . We will update it on our project and test is asap. The AP feature will also be very helpful.

mlubinsky, 2018-07-23:
How to get this patch https://github.com/ARMmbed/mbed-os/pull/7579
into my project, which is a fork of
 https://github.com/armmbed/ref-wem-firmware
? 
I am using mbed-cli. I need to test it ASAP.

screamerbg, 2018-09-10:
You can do 
```
$ cd <your project>
$ cd mbed-os
$ mbed update mbed-os-5.10.0-rc1
```

screamerbg, 2018-09-26:
This is now released with mbed OS 5.10.0. Just use `mbed update mbed-os-5.10.0`.

@dmelo Please try the new release and close the issue if it's resolved for you.

0xc0170, 2018-09-26:
I'll close it as resolved. Please reopen with an update if 5.10.0 does not fix it for you

closed 2018-09-26 by 0xc0170