ARMmbed/mbed-os #6129: build fail, L6218E : Undefined symbol $Super$$main in mbed_boot.o

phanirajkiran, 2018-02-18:
### Description
- Type: Question
- Related issue: `#build fail, L6218E : Undefined symbol $Super$$main in mbed_boot.o`
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
ANY TARGET BUILD FAIL IN MBED-CLI
**Toolchain:**
ARM

**Toolchain version:**
5.4.1 20160919

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
569159b784f7 (master)
https://github.com/ARMmbed/mbed-os#569159b784f7
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**
on Latest master branch sync from mbed-cli (1.4.0 version), fire build for example 

$ mbed compile -t ARM -m K64F -c -v

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question
Hello,
For any target, build error from the master branch (https://github.com/ARMmbed/mbed-os#569159b784f7).

Error: L6218E:Undefined symbol $Super$$main (referred from {HEAD_NODE/BUILD/TARGET_NAME/ARM/mbed-os/rtos/TARGET_CORTEX/mbed_boot.o}

Regards
Phanirajkiran
 
**How to?**

Please let me know how to resolve the issue. I understand that its trying to link a library & In application layer main() is missing. So which corresponding import i need to add in script so that this issue can be suppressed.

phanirajkiran, 2018-02-18:
Hey 

Got Fixed, by importing a sample and changing the compiler.

closed 2018-02-18 by phanirajkiran

HarishkumarBringiramakannu, 2018-06-20:
Hey @phanirajkiran , 
How did you fix this error ? Because I am getting the same error, but I am using online compiler.
Thank you in Advance

phanirajkiran, 2018-06-20:
Harish,
Send me build command n build log wat u did till now. It's simple I'll help
u.
Send me the MCU if it's geeko it's hard to find. Take help/ ref of freedomK
Cheers
Phani

HarishkumarBringiramakannu, 2018-06-20:
Hi @phanirajkiran ,
I am searching a good library for DS18B20 temperature sensor. But whatever I see, there is some error in header file. I am using STM32L072CZ microcontroller. If you have a good one, you can share with me. I am just checking with the library and I stopped with my main file. Even the libraries are not compiling properly.
Though I will attach the link here what I am using https://os.mbed.com/users/richardlane/code/DS18B20_1wire/
It shows some error in CRC, so if I comment those two lines. It shows this error.
If you can, please help me through this.

phanirajkiran, 2018-06-20:
Harish
I asked you for build log, r send screen shot
Thanks
Phani

HarishkumarBringiramakannu, 2018-06-20:
Hey @phanirajkiran ,
I am using online compiler where to see the build log ?

phanirajkiran, 2018-06-20:
Best is to make mbed-cli. To get inside mbed

HarishkumarBringiramakannu, 2018-06-20:
Hey @phanirajkiran ,
okay once I download mbed cli. I will get back to you in few minutes.