ARMmbed/mbed-os #6133: TCPServer only accepts first N connections

pauluap, 2018-02-19:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
DISCO_F769

**Toolchain:**
GCC_ARM

**Toolchain version:**
6.3.1.20170125

**mbed-os sha:**
5.7.5 tag

**Expected behavior**
The TCP server allows clients to connect and disconnect multiple times.

**Actual behavior**
The server callback is only called up to the number of listen backlog.  Once the connection is closed, the socket callback is never called again

**Steps to reproduce**
Create a TCPServer and call TCPServer::listen() with any number of backlog items.  Attempt any combination of connections (sequentially, multiple open, etc).  Once the number of backlog items has been reached, observe that any further attempts to connect will fail and the TCPServer socket callback is never called.

Here is a Wireshark entry for a single connection for a single request packet.  Then another connection is attempted and the lwip lower layer rejects the connection.

 ```
     1 0.000000000    Stmicroe_2a:00:2c     Broadcast             ARP      60     Who has 192.168.2.121? Tell 0.0.0.0
      2 0.462498427    Stmicroe_2a:00:2c     Broadcast             ARP      60     Who has 192.168.2.121? Tell 0.0.0.0
      3 0.462513603    Stmicroe_2a:00:2c     Broadcast             ARP      60     Who has 192.168.2.121? Tell 0.0.0.0
      4 0.962506289    Stmicroe_2a:00:2c     Broadcast             ARP      60     Gratuitous ARP for 192.168.2.121 (Request)
      5 1.462498395    Stmicroe_2a:00:2c     Broadcast             ARP      60     Gratuitous ARP for 192.168.2.121 (Request)
      6 2.462516536    Stmicroe_2a:00:2c     Broadcast             ARP      60     Gratuitous ARP for 192.168.2.121 (Request)
      7 3.462522534    Stmicroe_2a:00:2c     Broadcast             ARP      60     Gratuitous ARP for 192.168.2.121 (Request)
      8 11.657405660   192.168.2.195         192.168.2.121         TCP      74     36520 → 502 [SYN] Seq=0 Win=29200 Len=0 MSS=1460 SACK_PERM=1 TSval=66917205 TSecr=0 WS=128
      9 11.657627481   Stmicroe_2a:00:2c     Broadcast             ARP      60     Who has 192.168.2.195? Tell 192.168.2.121
     10 11.657650129   Tp-LinkT_01:85:2a     Stmicroe_2a:00:2c     ARP      42     192.168.2.195 is at 64:70:02:01:85:2a
     11 11.657751701   192.168.2.121         192.168.2.195         TCP      60     502 → 36520 [SYN, ACK] Seq=0 Ack=1 Win=24000 Len=0 MSS=1460
     12 11.657788543   192.168.2.195         192.168.2.121         TCP      54     36520 → 502 [ACK] Seq=1 Ack=1 Win=29200 Len=0
     13 11.657848128   192.168.2.195         192.168.2.121         Modbus/TCP 66        Query: Trans:     1; Unit:   1, Func:   4: Read Input Registers
     14 11.664707271   192.168.2.121         192.168.2.195         Modbus/TCP 207    Response: Trans:     1; Unit:   1, Func:   4: Read Input Registers
     15 11.664731697   192.168.2.195         192.168.2.121         TCP      54     36520 → 502 [ACK] Seq=13 Ack=154 Win=29047 Len=0
     16 11.664968649   192.168.2.195         192.168.2.121         TCP      54     36520 → 502 [FIN, ACK] Seq=13 Ack=154 Win=29047 Len=0
     17 11.665119478   192.168.2.121         192.168.2.195         TCP      60     502 → 36520 [ACK] Seq=154 Ack=14 Win=23987 Len=0
     18 11.668190273   192.168.2.121         192.168.2.195         TCP      60     502 → 36520 [FIN, ACK] Seq=154 Ack=14 Win=23987 Len=0
     19 11.668219584   192.168.2.195         192.168.2.121         TCP      54     36520 → 502 [ACK] Seq=14 Ack=155 Win=29047 Len=0
     20 18.056859860   192.168.2.195         192.168.2.121         TCP      74     36522 → 502 [SYN] Seq=0 Win=29200 Len=0 MSS=1460 SACK_PERM=1 TSval=66918805 TSecr=0 WS=128
     21 18.057097916   192.168.2.121         192.168.2.195         TCP      60     502 → 36522 [SYN, ACK] Seq=0 Ack=1 Win=24000 Len=0 MSS=1460
     22 18.057120979   192.168.2.195         192.168.2.121         TCP      54     36522 → 502 [ACK] Seq=1 Ack=1 Win=29200 Len=0
     23 18.057156665   192.168.2.195         192.168.2.121         Modbus/TCP 66        Query: Trans:     1; Unit:   2, Func:   4: Read Input Registers
     24 18.255155263   192.168.2.195         192.168.2.121         TCP      66     [TCP Retransmission] 36522 → 502 [PSH, ACK] Seq=1 Ack=1 Win=29200 Len=12
     25 18.255348380   192.168.2.121         192.168.2.195         TCP      60     502 → 36522 [ACK] Seq=1 Ack=13 Win=23988 Len=0
     26 20.059238464   192.168.2.195         192.168.2.121         TCP      54     36522 → 502 [FIN, ACK] Seq=13 Ack=1 Win=29200 Len=0
     27 20.059415875   192.168.2.121         192.168.2.195         TCP      60     502 → 36522 [ACK] Seq=1 Ack=14 Win=23987 Len=0
```

The same thing is observed when I create a TCPServer with a backlog of 5 connections.  I'm able to open five connections (or one connection five times), the sixth attempt fails.

I have stepped through my code and I can say that I'm calling close() on the socket that I receive from TCPServer::accept().  Did I miss a procedure for returning a socket to the TCPServer listening pool?

Thanks

0xc0170, 2018-02-19:
@kjbracey-arm @mikaleppanen  @ARMmbed/team-st-mcd Please review

kjbracey-arm, 2018-02-19:
The backlog limit should only be applying to incoming connections that you've yet to accept. If you've accepted it, you should have cleared the backlog, and made the stack ready for the next connection.

That trace looks like the second connection was accepted by lwIP, but not by the application (because you weren't told about it?) - it's completed the syn and fin handshake and accepted the data, but not sent anything or closed itself.

One slightly half-hearted suggestion - not sure this is what you're seeing. If you're running asynchronously, make sure your handler calls accept() repeatedly until you get a "would block" error before waiting for the next signal. You're not guaranteed one signal per incoming connection attempt (or data) - always read/accept as much as you can. Only wait for the signal callback after you attempted a socket call and got a "would block" error.

pauluap, 2018-02-19:
Thank you!  I was pretty sure that I was hitting TCPServer::accept() until
I got NSAPI_ERROR_WOULD_BLOCK.  Upon reflection, I decided to try calling
TCPServer::accept() right after closing the TCPSocket() that was previously
completed.  So your guess of "you weren't told about it" seems to be
correct.

I am absolutely sure, however, that I do not receive a TCPServer signal and
"ignore" it - I have a print statement that kicks on every call to the
signal handler.  So it appears to me that once a TCPServer socket listen
backlog has been exhausted, it will not get any more signals until accept()
has been called once again.  I chose to do it after I've closed a socket
because I don't want to deal with either NSAPI_ERROR_NO_SOCKET or an
unexpected established connection for more than my backlogged items.  Can
you confirm that's expected behavior?  I totally get the "not guaranteed a
signal per attempt" but that closing a TCPSocket associated with a server
does not clear the "do not kick the signal handler mechanism on socket
activity" flag, and calling  TCPServer::accept() is the "clearing the
backlog and made the stack ready for the next connection" process?

What is the Clear Backlog And Make Stack Ready procedure?

Thank you for the assistance, I believe that the original issue has been
resolved and the remainder is followup clarifications and recommendations

kjbracey-arm, 2018-02-19:
There is a potential inherent race condition if you do not process until you've confirmed "no more data / connections". The signal generation will be dependent on the network stack, and the only thing you can be certain of is that it must signal the "nothing to do -> something to do" transition. It may not signal "even more to do".  So I would say that you must always ensure you've confirmed you've been told "nothing to do" to ensure you will definitely get a (or another) signal. 

The processing/accepted socket and the listening socket are (I believe) independent. So after accepting, what you do on the accepted socket shouldn't have any impact on the listening socket.

There is a suppression mechanism per-socket in the lwip glue - after signalling you once, it won't signal you again on that socket until you've made at least one socket call on it. The sequence "signal for incoming connection", "accept" should clear the suppression flag for the listening socket, so you should get another signal as long as the next connection came in after that accept. But as that's a race between the second connection and the first accept, you can't rely on it.

Therefore you must definitely try accepting immediately after accepting then closing the last connection, if you have that sort of single-threaded design.

I find the best way to think of this is write your code as if it was blocking code. You would naturally write a loop that did `listen() for (;;) { accept(); process(); close (); }`. You need to put the "waiting" code only where it would block - ie during the accept when there's nothing available, not between the close and accept.

Something does feel a little off here - I wouldn't expect the behaviour to change after "backlog" connections. In the simple one-at-a-time connection case, where you're not hitting the race, I would expect you to be getting a wake-up signal, even though you didn't confirm there were no pending connections.

pauluap, 2018-02-19:
Allright.  What you say makes sense, so I set up an minimal working example.  I think that I now see where I went wrong.

The code I used to test this is:

```
#include <iostream>

#include "mbed.h"
#include "EthernetInterface.h"

EthernetInterface net;
EventQueue queue;
TCPServer srv;
TCPSocket sock;

void SrvSignal()
{
    nsapi_error_t res = srv.accept(&sock);
    std::cout << clock() << ": Accept attempt resulted in: " << res << "\n";
    if (NSAPI_ERROR_OK == res)
    {
        sock.close();
    }
}

int main() {
    std::cout << "Starting at: " << clock() << "\n";
    net.connect();

    std::cout << "IP address is: " << net.get_ip_address() << "\n";

    srv.open(&net);
    srv.bind(502);
    srv.set_blocking(false);
    srv.sigio(queue.event(SrvSignal));
    srv.listen();
    queue.dispatch_forever();
}
```

I tried a couple of connection attempts, and the output was:

```
Starting at: 0
IP address is: 192.168.2.121
2507: Accept attempt resulted in: 0
2511: Accept attempt resulted in: -3001
3927: Accept attempt resulted in: 0
3931: Accept attempt resulted in: -3001
```

So in my case, I had a maximum connection limit.  When I reach my self-imposed connection maximum, I do an accept, then the TCPServer socket would immediately raise another signal, expecting me to call it and receive an NSAPI_ERROR_WOULD_BLOCK result.  But at that point, I'm not supposed to have any more sockets available (may even be at the system configured maximum of `MBED_CONF_LWIP_TCP_SOCKET_MAX`) so I'm unable to call accept() in order to clear the blocking flag.

That does explain why I fall off the cliff when I hit my limit and calling accept() on my next socket closure makes the signaling work again.

After thinking about it, this behavior does seem correct to me, it's pretty much exactly what you said in your loop example, after closing, I need to go hit up accept() and not wait between close() and accept().

I think I'm good on this one, thank you for the valuable information!

kjbracey-arm, 2018-02-19:
I'm wondering if this is to do with the "negative" events that lwIP signals. It oddly has upcalls for both "more stuff available" and "less stuff available". Calling read (and I assume accept) generates a "less stuff" upcall, and in some cases this isn't filtered out, so generates a sigio, thus setting the suppression flag. And that can effectively simulate the possibly real race - you get this spurious sigio "during" the read/accept, which means you then don't get a later real one if you don't make sure to make another socket call.

My recollection is we did try filtering tnhese more, but found that it broke some code that was somehow relying on the spurious events.

Anyway, if you have any self-imposed limits that stop you processing, it's up to you do proceed on the sockets yourself when that condition has cleared. The stack's probably sitting there twiddling its thumbs waiting for you to do something.

I'm not quite sure how the lwIP socket maximum relates to incoming TCP connections. My assumption would be that the embryonic connection consumes a socket - it wouldn't handshake with the peer if it didn't have a socket PCB free, so it must always be possible to accept() if it's done a handshake. I expect the accept just gives you an identifier for the socket structure that's already been assigned to process the connection.

Anyway, okay to close this?

pauluap, 2018-02-19:
Yeah, that's my understanding as well in regard to the socket construction during the connect call.  I'm not blocked anymore, so thanks! back to trying to figured out #6130

closed 2018-02-19 by pauluap