ARMmbed/mbed-os #6137: Printing `uint64_t` does not work

drahnr, 2018-02-19:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM

**mbed-os sha:**
4.7.4 + 2 patches
commit caeaa49d68c67ee00275cece10cd88e0ed0f6ed3 (tag: mbed-os-5.7.4, origin/mbed-os-5.7)

**Expected behavior**
Printf should format uint64_t properly

**Actual behavior**
Fails, screws over part of the stack and does not print properly.

**Steps to reproduce**

```
#include <inttypes.h>
const us_timestamp_t now = 10000000;
// any of the below blow up
printf("%" PRIu64 " ", now);
printf("%8llu ", now);
printf("%llu ", now);
```

0xc0170, 2018-02-19:
Isn't this because `__STDC_FORMAT_MACROS` is not defined ?

drahnr, 2018-02-19:
It still prints `lu ` instead of the actual timestamp.

pan-, 2018-02-19:
@drahnr Are you compiling a C or a C++ file ? We do use C++98 and C99; obviously C++ code shouldn't be aware of C99 facilities but vendor leak most of it and it's used all over the place in the code base (event if not valid). 

```C++
#include <inttypes.h>
const us_timestamp_t now = 10000000; // or 0
// any of the below blow up
printf("%" PRIu64 " \n", now);
```

Compiled in a C file should print your timestamp. If it doesn't then could you tell use which version of GCC is used ?

drahnr, 2018-02-19:
I am using C++11 in this codebase, but I think the focus here is wrong. This is not about `PRIu64` not being defined/being defined, this is about `%llu` not yielding the output expected from `printf`!

```
printf("%8llu \n", now);
printf("%llu \n", now);
```

0xc0170, 2018-02-20:
This target should use full lib, thus newlib (not nano) - should support 64bit stdin/out. This bug is related to the library.

> arm-none-eabi-gcc 7.1.0

Note, mbed OS supports 6.x currently.

drahnr, 2018-02-21:
Indeed,         `"default_lib" : "small",` in `custom_targets.json` caused this.

0xc0170, 2018-02-22:
> Indeed, "default_lib" : "small", in custom_targets.json caused this.

Does this mean this can be closed?

closed 2018-02-22 by drahnr