ARMmbed/mbed-os #6150: Cellular Generic Driver unit tests fail

RobMeades, 2018-02-21:
### Description
- Type: Bug
- Related issue: #6032 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
`UBLOX_C027`/`UBLOX_C030_U201`/any cellular

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
636ced8ed

**Expected behavior**

The tests in `mbed-os-features-netsocket-cellular-generic_modem_driver-tests-unit_tests-default` should pass on a cellular platform, specifically `UBLOX_C027` and `UBLOX_C030_U201`.

**Actual behavior**
The first test passes but all subsequent tests fail.  The connection actually succeeds ('cos I can see the IP address and DNS information being printed out) but the `connect()` call does not return, instead it times out, returning error -3004 `NSAPI_ERROR_NO_CONNECTION`.  To be completely sure, I then modifed the tests so that the second test is the first test run once more.  A log of the failure behaviour is attached in
[generic_modem_driver_test.txt](https://github.com/ARMmbed/mbed-os/files/1743573/generic_modem_driver_test.txt) (note that I'm bouncing off my own echo server in that test run so that I can see the echo requests arriving (or not)).

If I switch to the commit before #6032 was merged (5a7482667e86f231c51efad959e7bbed228fdd2e) then the tests pass, so I suspect that's caused the problem.

**Steps to reproduce**
Get a cellular platform, e.g. `UBLOX_C027` or `UBLOX_C030_U201` with a valid SIM and run the following:

`mbed test -v -n mbed-os-features-netsocket-cellular-generic_modem_driver-tests-unit_tests-default`

@hasnainvirk, @kjbracey-arm FYI.

RobMeades, 2018-02-21:
For completeness, I can confirm that the commit 282f3c9 (#6032) does show the problem while the commit before that (5a74826) does not.

MarceloSalazar, 2018-04-19:
Closing as a fix is confirmed to be in place.

closed 2018-04-19 by MarceloSalazar