ARMmbed/mbed-os #6158: STM32F429: "usb_thread could not read dev descr" error in USB host keyboard

guest253456, 2018-02-22:
### Description
- Type: Bug
---------------------------------------------------------------
## Bug

**Target**
NUCLEO-STM32F429

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
caeaa49

mbed's USB host keyboard is not working, i'm getting the following error when i try to use the usb host keyboard
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_ERR: ../USBHost/USBHost/USBHost.cpp:149]usb_thread could not read dev descr
>[USB_INFO: ../USBHost/USBHost/USBHost.cpp:177]New device connected: 0x2000b414 [hub: 0 - port: 1]

0xc0170, 2018-02-22:
@ARMmbed/team-st-mcd Please review

timkrief, 2018-02-23:
I have the same problem here. I can't connect a keyboard on an usb hub connected to **STM32F429I-DISC1** ... The library seams to not work properly.

jeremydufour, 2018-02-23:
@adbridge any ideas on this issue ?

adbridge, 2018-02-23:
@jeremydufour We are waiting on ST to take a look.

bcostm, 2018-02-24:
No idea why it does not work on the NUCLEO_F429ZI, we'll have to look at it.

But concerning the support of USBHost on the DISCO_F429ZI is has not been enabled due to unresolved issue (see PR #5617). No progress since that time on this board.

Piroro-hs, 2018-03-04:
I have the same problem with NUCLEO-F446RE board.
Each time I connect a USB device, I get 10 USB_ERR and USB_INFO with same id. Also, when I reset the board, it says mouse started.

jeremydufour, 2018-04-08:
@bcostm I added support for DISCO-F429ZI. My commit is here 859c607f30bd989ab6acde333980daea791d1827

I need your approval before opening a pull request

bcostm, 2018-04-09:
Thanks ! I'll have a look and let you know.

timkrief, 2018-04-09:
Thank you @jeremydufour I hope ARMmbed will use this as soon as possible!

bcostm, 2018-04-09:
@jeremydufour So, I checked your commit and it is almost the same as I did (and removed) in my commit https://github.com/ARMmbed/mbed-os/pull/5617/commits/c621c688c4777b5bef65ea00d4bfb537d547cf12. Only difference is that in your case you didn't configured the VBUS pin.

Anyway it didn't work for me. I test it using the `features\unsupported\tests\usb\host\mouse` test.

Can you tell me the board name/version you have ? The state of the solder bridges SB18, SB19 and SB20. And finally can you test the USB Mouse test code ? Thanks.

jeremydufour, 2018-04-16:
@bcostm Sorry for the late reply. I have a STM32F429I-DISC1 board. I can't test the solder bridges right now but I keep you updated for the next two weeks. I will try soon the USB Mouse test code as well.

0xc0170, 2018-06-19:
@jeremydufour  Any update?

@ARMmbed/team-st-mcd What is the status for this issue?

screamerbg, 2018-09-10:
@jeremydufour Any update on this?

jeremydufour, 2018-09-24:
Hi everyone ! I have been working on the BlueNRG2 SoC recently. I don't have any updates on this issue but I will send updates when I have some time.

bcostm, 2018-09-24:
No update also on our side with the actual API.

The goal now is to work on the new USB API (branch https://github.com/ARMmbed/mbed-os/tree/feature-hal-spec-usb-device) only. But nothing done yet on this board AFAIK.

jeromecoutant, 2019-02-20:
It seems there is no more pending action.
Can we close this old issue ?

closed 2019-02-20 by Arjun765