ARMmbed/mbed-os #6174: memap ImportError

hosse005, 2018-02-22:
### Description
- Type: Question
- Priority: Minor

---------------------------------------------------------------
## Question
Anybody noticing memap.py ImportError failures with mbed-os-5.7.5?

**Target**
Not target specific

**Toolchain:**
Not toolchain specific

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
569159b

**Expected behavior**
$ python memap.py displays help screen

**Actual behavior**
Traceback (most recent call last):
  File "memap.py", line 16, in <module>
    from tools.arm_pack_manager import Cache
ImportError: No module named tools.arm_pack_manager

**Steps to reproduce**
$ python memap.py

0xc0170, 2018-02-23:
@theotherjimmy Please review

hosse005, 2018-02-23:
Not a huge issue for me, just noticed that it was working with mbed-os-5.7.4 and wasn't sure if it was a known issue or intended.

theotherjimmy, 2018-02-23:
@0xc0170 I think it may have worked before. I'm not sure that we will continue to have that entry point work. It's not an official entry point, as it has no Mbed CLI verb associated with it.

0xc0170, 2018-02-26:
> @0xc0170 I think it may have worked before. I'm not sure that we will continue to have that entry point work. It's not an official entry point, as it has no Mbed CLI verb associated with it.

If working as intended (use mbed-cli), we can close the issue?

hosse005, 2018-02-26:
@0xc0170 , that's fine with me.

closed 2018-06-19 by 0xc0170