ARMmbed/mbed-os #6180: KL27Z "Unsupported warning"

ManDeJan, 2018-02-23:
I am working with mbed OS 5.7.5 and the KL27Z board,
I get an error if I want to compile that says board not supported
It seems the file [/targets/targets.json](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json) is out dated, I added a "5" to the release versions of KL27Z and everything worked fine. Is there a reason it is not yet updated?

0xc0170, 2018-02-23:
The target was enabled only for mbed 2 (as you noticed).

@MarceloSalazar Can you review please?

MarceloSalazar, 2018-02-26:
@maclobdell please review

maclobdell, 2018-03-27:
My apologies, I just realized that I had not commented on this.  During testing, this platform was found to run out of memory with some Mbed OS 5 tests with GCC_ARM compiler.  It worked fine for IAR and Arm compilers, which are optimized more for small memory footprint.  The KL27Z passed tests with Mbed 2 configuration without RTOS, which uses less memory.  This is why the platform was marked as Mbed Enabled for Mbed 2 only.

ManDeJan, 2018-03-27:
Hi, thanks for the response. After editing targets.json and enabling Mbed OS 5 it worked fine for me with the gcc-arm compiler toolchain. Though I needed to turn on -Os optimizations. Building the following code for the KL25Z target gives a 45K binary. Building it for the KL27Z gives a 53K binary.

```cpp
#include "mbed.h"

int main() {
    DigitalOut led_pin(LED1);
    while (true) {
        led_pin = 1;
        wait_ms(500);
        led_pin = 0;
        wait_ms(500);
    }
}
```
These are very similar chips and such a simple program as above shouldn't result in a binary 8 kilobytes larger. After inspecting both binarys I found the KL27Z target contains an additional 115 symbols, I am suspecting most of these functions are simply not marked as inline and are not necessary. Here are the symbols for both projects: https://pastebin.com/3jgPRqbi (kl27z) https://pastebin.com/vBV8UFxm (kl25z).

0xc0170, 2018-06-19:
> My apologies, I just realized that I had not commented on this. During testing, this platform was found to run out of memory with some Mbed OS 5 tests with GCC_ARM compiler. It worked fine for IAR [...]

This issue will be closed. The question was answered.

@ManDeJan If you still have an issue or question regarding KL27Z , please create a new issue

closed 2018-06-19 by 0xc0170