ARMmbed/mbed-os #6181: Greentea tests failing with mts mdot f411re & IAR combination

kimlep01, 2018-02-23:
### Description
- Type: Bug
- Priority: Major
---------------------------------------------------------------
## Bug

**Target**
MTS_MDOT_F411RE

**Toolchain:**
IAR

**Toolchain version:**
7.80.1

**mbed-cli version:** 
1.4.0

**mbed-os sha:**
1d1eef4 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #6160 from TeemuKultala/status_callback_fix

**DAPLink version:**
0221
**Expected behavior**
Greentea tests pass when compiled with IAR 7.80.1 in Linux.

**Actual behavior**
Even the most basic 'mbed-os-tests-integration-basic' fails due to "SYNC_FAILED". The same test passes without issues with ARM, GCC_ARM and ARMC6.

**Steps to reproduce**
Install IAR 7.80.1 compiler in Linux (Ubuntu 16.04)
Use one example repository
Checkout latest mbed-os master
remove main.cpp from example
mbed test --compile -n mbed-os-tests-integration-basic -m MTS_MDOT_F411RE -t IAR
mbedgt -n mbed-os-tests-integration-basic -V

0xc0170, 2018-02-23:
Only for IAR

Can you test simple blinky and debug if it works? Possibly even a test case with serial, some drivers test we have, and comment out test prints, to be able to compile and run it.

This is target based on F411, we might want to also test nucleo f411 target to minimize the code that is different from STM F4 targets

kimlep01, 2018-02-23:
Blinky does not blink. I also tested mbed-os-TESTS-mbed_drivers* and they all fail to sync after reset. Again no issues with other compilers.

0xc0170, 2018-02-26:
@ARMmbed/team-multitech Please review

klaas019, 2018-02-26:
How did you manage to install IAR 7.80.1 compiler in Linux (Ubuntu 16.04)? Wine?

kimlep01, 2018-02-27:
I am using the command line compiler in Linux. Tested also in Windows 10 with 7.80.4 with the same results.

0xc0170, 2018-03-09:
#6311 should address this, @kimlep01  please review

kimlep01, 2018-03-09:
All Greentea tests pass now with IAR, closing defect.

PR #6311 (2018-03-15): mts_mdot_f411re: Fix for Multi-Tech mDot IAR linker script
### Description
Fix for applications targeting the mts_mdot_f411re target with the IAR compiler.  Need the reset vector table to also have a 64 kB offset to compensate for the presence of a 64 kB bootloader.

Verified by building and running the mbed-os-example-blinky app successfully with the following environment:
* IAR EWARM 7.80.1
* mbed-cli 1.3.0

Fixes #6181 

### Pull request type
- [x] Fix
- [ ] Refactor
- [ ] New target
- [ ] Feature
- [ ] Breaking change
closed 2018-03-09 by kimlep01