ARMmbed/mbed-os #6184: ST's USBDevice library not working properly on STM32F429I

jeremydufour, 2018-02-23:
### Description
- Type: Bug
- Priority: Major
---------------------------------------------------------------
## Bug

**Target**
STM32F429I-DISC1

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
5.4.1

**mbed-cli version:** 
mbed-cli not used in this project

**mbed-os sha:**
mbed-os version 5.6.6

**Expected behavior**
The STM32 board has to emulate a USB keyboard compatible with any USB port

**Actual behavior**
The code is stuck and doesn't go further, I have some prints that are not displayed on the terminal

**Steps to reproduce**

The USBDevice library has to be fixed. The code is probably stuck at the USBDevice::connect(bool blocking) function. I use the USBDevice library made by ST available here: https://os.mbed.com/teams/ST/code/USBDEVICE/#d9c7334e2183

This is the main.cpp file I use for this project:

`
#include "mbed.h"
#include "USBKeyboard.h"

DigitalOut led_green(LED3);
DigitalOut led_red(LED4);

USBKeyboard key;

InterruptIn user_button(USER_BUTTON);

bool button_state;
int carac;

void button_pressed()
{
    button_state=true;
}

void button_released()
{
    button_state=false;
}

//************Additionnal test features

int main(){
    while(1){
        printf("Test d'emulation de clavier\n");
        led_green=1;
        led_red=1;
        user_button.rise(&button_pressed);
        user_button.fall(&button_released);
        carac=0x0b; // stands for h
        carac=0x28; // stands for \n or enter
        if(button_state) {
           printf("tentative d'ecriture\n");
           led_green=1;
           led_red=0;
        }
        else{
            led_green=0;
            led_red=1;

        } 
        wait(1);
    }
}`

0xc0170, 2018-02-23:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-02-24:
I see that you use the library available in mbed.com and not the files available in GitHub mbed-os/feature/unsupported folder. I am not sure if the library on mbed.com is up to date. We'll have to verify it. Concerning the support of USBHost on the DISCO_F429ZI, it has not been enabled on this platform due to unresolved issue. See PR #5617. No progress since that time on this board. Community help is welcome.

jeremydufour, 2018-02-25:
Thank you for the advice, I used the USBDevice library from GitHub mbed-os/feature/unsupported folder and I could connect the STM32 to my laptop and write some characters with it.

jeromecoutant, 2018-03-14:
Hi @jeremydufour 
Maybe we can close this issue ?
Thx

jeremydufour, 2018-03-15:
@jeromecoutant of course we can close it. I don't have issues on this library anymore. 
Thank you for the support you gave me.

closed 2018-03-15 by jeremydufour