ARMmbed/mbed-os #6187: Minor change of SPI interface [API breaking]

drahnr, 2018-02-23:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement | Question
- Priority: Minor

---------------------------------------------------------------

## Question

using SPI the types do not make much sense.

The idea is that things are bytes that are written, I am not 100% sure if `char`is guaranteed to be unsigned. At least as of `C11` this is implementation specific.

So should there not be either be made usage of `uint8_t ` (as defined in <`cinttypes>` or `<cstdint>`)  or define a custom `byte_t`.

`virtual int write(const char *tx_buffer, int tx_length, char *rx_buffer, int rx_length);`

The API of `virtual int write(int value);` is not very idiomatic either, the argument should also be an `uint8_t` since that is what is actually transfered, the return value is a mingled version of an error code and a return code.

proposal (this is an API breaking change):

`virtual int write(const uint8_t *tx_buffer, size_t tx_length, uint8_t *rx_buffer, size_t rx_length);`
`virtual int write(uint8_t value, uint8_t *ret);`

I'd like to know what you think about this or what was the reasoning to use the above?

deepikabhavnani, 2018-08-27:
> I'd like to know what you think about this or what was the reasoning to use the above?

Work on these lines is going on at the HAL layer, once low level API's are updated we will update the driver as well. Refer https://github.com/ithinuel/mbed-os/blob/rfc-spi-overhaul/hal/rfcs/0000-spi-overhaul.md for updated API's

evedon, 2019-05-23:
HAL layer was updated. 
Closing this question as it is rather old. Please reopen if you need additional assistance.

closed 2019-05-23 by evedon