ARMmbed/mbed-os #6194: Tickers are not halted when getting into error()

amq, 2018-02-23:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
Tested on EFM32PG12 with LowPowerTicker

**Toolchain:**
GCC_ARM

**Toolchain version:**
6-2017-q2-update

**mbed-cli version:** 
1.3.0

**mbed-os sha:**
caeaa49d68c67ee00275cece10cd88e0ed0f6ed3

**Expected behavior**

When the system goes into `error()`, all threads and tickers should be stopped. Relevant for watchdog feeding / kicking.

**Actual behavior**

Tickers continue to execute and feed watchdog, preventing it from resetting the board.

If this is expected, then the Watchdog API docs should clearly state that `kick()` has to be called from a thread context.

**Steps to reproduce**

```
#include "mbed.h"

int main() {
#ifdef DEVICE_LOWPOWERTIMER
    static LowPowerTicker statusTicker;
#else
    static Ticker statusTicker;
#endif

    const float STATUS_BLINK_FREQUENCY_SEC = 0.5f;
    static DigitalOut led(LED_GREEN);
    void (*watchdogRefreshIsr)() = []() {
        led = !led;
        // target-specific feed function
    };

    statusTicker.attach(watchdogRefreshIsr, STATUS_BLINK_FREQUENCY_SEC);
}
```

0xc0170, 2018-02-26:
@scartmell-arm Please review

scartmell-arm, 2018-02-26:
That seems expected, if you're using something that still runs in a crashed state to kick a watchdog, the watchdog isn't going to trigger a reset.

amq, 2018-02-26:
Right, my main surprise wasn't that the watchdog wasn't triggered, but that the ticker and isr continued to run.

amq, 2018-02-27:
@scartmell-arm actually, I've just found out that even a thread can continue running after we've hit `error()`. In this case I'm not sure how to make watchdog work at all.

daniel-cesarini-tridonic-com, 2018-09-24:
Stupid question: why is there the need to have the watchdog resetting the device/system when ending in error / mbed_die? Is it not possible to issue a Software Reset in the context of error / mbed_die (which should already be the case)?

amq, 2018-09-24:
@daniel-cesarini-tridonic-com yes, that's what I ended up doing

aashishc1988, 2018-10-02:
@amq I started looking at this issue. I used the program you provided in the example, and called MBED_ERROR to create an error just so that I can replicate the behavior you are seeing. I called
```
MBED_ERROR1(MBED_MAKE_ERROR(MBED_MODULE_KERNEL, MBED_ERROR_OPERATION_PROHIBITED), "Kernel error", 0xDEADDEAD );
```
And as expected I got

```
++ MbedOS Error Info ++
Error Status: 0x80020110 Code: 272 Module: 2
Error Message: Kernel error
Location: 0x78B9
Error Value: 0xDEADDEAD
Current Thread: Id: 0x20002138 Entry: 0x7D51 StackSize: 0x1000 StackMem: 0x20001138 SP: 0x20002090
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020110
-- MbedOS Error Info --
```
I modified your program to create multiple threads and tickers, and as soon as the error was encountered, the threads and ticker stopped running. Is there something I am missing? 
Also, just putting it out there in case this might be an issue, the LED you are using in your example, and the LED used by mbed_die is same, `LED1` , so it might seem that the thread is still running, but its the mbed die running that LED at a different frequency.

Can you help me reproduce this issue?

aashishc1988, 2018-10-17:
@amq I am going to close this issue as I do not think this is a problem. We do not run tickers and threads once we encounter mbed_error. It is, however, possible that the mbed os version you used might not have that change in where we disable all interrupts once we hit mbed_error. Please feel free to reopen the issue if you see that I missed something here.

closed 2018-10-17 by studavekar