ARMmbed/mbed-os #6196: Support for handling BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST for NRF52 devices in BLE API.

yogeshk19, 2018-02-23:
### Description
- Type:  Question
- Related issue: `#4893,#3509`
- Priority: Major 
-----------------------------------------------------------------
## Question

**How to handle BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST for NRF52 devices in BLE API? Currently there is no available mechanism to handle this event using the BLE API, as this seems to tie in to the low power mode operation for NRF52 devices. Currently using two NRF52 devices as peripheral and master and when the peripheral requests an update to the connection parameters the master has no mechanism in the mbed BLE API to handle this event BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST and to accept the updated connection request. What is the easiest way to achieve this?**

Thanks,
Yogesh

0xc0170, 2018-02-26:
@pan- please review

yogeshk19, 2018-03-08:
@pan-  Hey Vincent any thoughts if there is a plan to implement this in the BLE API or what is the best way to implement this?

Thanks,
Yogesh

pan-, 2018-03-12:
@yogeshk19 That's definitely missing in the API and should be patch. At least as default parameter it would make sense to accept connection parameters requests from peripherals because peripheral usually now better which parameters it needs.

Before we got a fix in master and a new API in place, here's the patch you need to accept connection parameter update requests: 

```diff
--- a/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF5/source/btle/btle.cpp
+++ b/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NRF5/source/btle/btle.cpp
@@ -292,6 +292,15 @@ static void btle_handler(ble_evt_t *p_ble_evt)
             break;
         }

+        case BLE_GAP_EVT_CONN_PARAM_UPDATE_REQUEST: {
+            Gap::Handle_t connection = p_ble_evt->evt.gap_evt.conn_handle;
+            const ble_gap_evt_conn_param_update_request_t *update_request =
+                &p_ble_evt->evt.gap_evt.params.conn_param_update_request;
+
+            sd_ble_gap_conn_param_update(connection, &update_request->conn_params);
+            break;
+        }
+
         default:
             break;
     }
```

yogeshk19, 2018-05-03:
@pan- Is there plan on integrating code into any upcoming releases of MBED OS?

Thanks,
Yogesh

donatieng, 2018-07-25:
Fixed by #7211

closed 2018-07-25 by 0xc0170