ARMmbed/mbed-os #6199: Why does default_idle_hook() lock deep sleep?

amq, 2018-02-23:
### Description
- Type: Question
- Related: https://github.com/ARMmbed/mbed-os/pull/5866

## Question

According to the documentation, sleep is managed automatically, including deep sleep. However from what I see, deep sleep is locked by `default_idle_hook()` for non-tickless targets

https://github.com/ARMmbed/mbed-os/blob/ce4d6c9f367031886b668c4fad1b0c2fd5006858/rtos/TARGET_CORTEX/mbed_rtx_idle.cpp#L220-L228

What is the reason for this? What is the expected way of entering deep sleep?

I know that it is possible to use a custom `default_idle_hook()`, but before I do, I would like to understand it more.

0xc0170, 2018-02-26:
This was intentionally added for backward compatibility. Previously only sleep was invoked, therefore deep sleep is locked to keep the same behavior. If you remove the deep sleep locking, some platform might potentially not wake up.

Does that answer your question?

amq, 2018-02-26:
How is deep sleep to be used? Is it just meant for tickless targets or is `Thread::attach_idle_hook(&sleep);` also a good approach?

0xc0170, 2018-02-26:
Yes, you can attach your own handler and overwrite the default behavior. Or enable tickless for your platform (there is feature sleep branch to actually address this, it contains specifications and tests, target additions will follow).

closed 2018-02-26 by amq