ARMmbed/mbed-os #6204: Should break_dispatch set a flag rather than incrementing?

pauluap, 2018-02-24:
I was inspecting the event queue implementation to see my intended usage was workable, and I came across this line.

https://github.com/ARMmbed/mbed-os/blob/569159b784f70feaa32ce226aaca896fb83452f7/events/equeue/equeue.c#L369

Later on,
https://github.com/ARMmbed/mbed-os/blob/569159b784f70feaa32ce226aaca896fb83452f7/events/equeue/equeue.c#L442

Since the event queue dispatch loop decrements `q->breaks` rather than setting it to 0, that effectively means that the event queue will break the next time `dispatch_forever()` is called if there has been more than one `break_dispatch()` call performed in the past.

My intended usage may lead to more than one break_dispatch call to the queue.  I want to know if there's any issue with my changing the code to set `q->breaks` to 1 rather than incrementing.

I would like the event queue to be broken when I am running it, I don't yet see the point of having subsequent calls to `dispatch_forever()` behave as if either `dispatch(0)` or `dispatch(-1)` is called depending on how many times `break_dispatch()` has been previously called.

0xc0170, 2018-02-26:
@geky @kjbracey-arm Please review

kjbracey-arm, 2018-02-26:
What you say makes sense to me - I'm always inclined to find counting signals like sempahores more often confusing than useful.

I believe in this API it is important that it be possible to cause an immediate exit from the next dispatch, to avoid a break/dispatch entry race. But I don't see the need for the counting. So I'd be inclined to have a simple boolean flag set by `break_dispatch`, and cleared as `dispatch` actually exits. (Maybe it should also be cleared when it exits by timeout?)

I guess @geky may have had a use case for the counting in mind though.

geky, 2018-02-27:
Makes sense to me too. I don't remember the original reason for break to be counting. Looking back on the history, originally break was not counting when first introduced. It was changed while optimizing for jitter, so it may have just been a shortsight on my part.

I'm open to changing it.

kjbracey-arm, 2018-02-28:
Then I suggest @pauluap submit a PR with a patch along the lines discussed - make it boolean. Welcome to the team :)

pauluap, 2018-02-28:
Done, and thanks!

PR #6238 (2018-03-15): Remove windup behavior from break_dispatch
Change break_dispatch from a counting semaphore which needs the same number of calls to dispatch_forever() as number of calls to break_dispatch() to a squashable flag which will only break once for any number of calls to break_dispatch()  between calls to dispatch_forever()

Resolves #6204

### Pull request type

- [ ] Fix
- [ ] Refactor
- [ ] New target
- [ ] Feature
- [x] Breaking change
closed 2018-03-15 by cmonr